echostack-org/echopype #825: Temporary directory deletion with `open_raw` and `offload_to_zarr=True`

b-reyes, 2022-10-04:
When we use `open_raw(offload_to_zarr=True)`, a temporary directory is created using `tempfile.TemporaryDirectory()`. Recently, I took a look at the path where these temporary directories are stored and I found that I had accumulated several of these directories and they were not deleted. These should be deleted once all references to the `TemporaryDirectory` are gone and this should happen when an `EchoData` object is deleted. 

From my preliminary investigation, these directories stick around because Jupyter Notebook is not properly releasing them. The issue does not happen when we run `open_raw(offload_to_zarr=True)` from a script. Jupyter notebook seems to track the `tempfile.TemporaryDirectory()` object. However, there is an issue with this tracking and the directories will persist, even if you restart or close the Jupyter notebook!

Additionally, I found that the name generated for the temporary directory is very general. This makes it hard to manually remove the temporary directories because you need to first look at the contents of the directory and make sure it was generated by echopype. Until we can take a serious look into this (or create an issue with Jupyter Notebook), I suggested that we at least add a prefix to these names, say `tempfile.TemporaryDirectory(prefix="echopype_")`. This will allow one to easily identify these directories and remove them.

leewujung, 2022-10-04:
@b-reyes and @leewujung just discussed this. We thought that we can circumvent this problem by the following:
- instead of using `tempfile` to create a temp directory, we create a subdirectory `SUB_NAME` under the default `temp_echopype_output` and put the temporary zarr files there.
- the zarr files will be named something like `temp_gf6f78dghjfdk.zarr` and get removed if users remove the `EchoData` object associated with it
- and because we know there may be problems in the object destructor, we can let users know that if these `temp_*.zarr` files are not removed automatically they can just remove the whole `SUB_NAME` directory.

b-reyes, 2022-10-07:
@leewujung if you think PR #832 adequately addressed this issue, please feel free to close it.

closed 2022-10-07 by leewujung