echostack-org/echopype #831: Develop tests for prov module, specially for source file name functionality

emiliom, 2022-10-05:
- From @lsetiawan in PR #829: I realized that we don't have a test for the [utils/prov.py](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/utils/prov.py), maybe this will be great for our testing creation session in the future :smile:
- From @emiliom in PR #814: While creating this PR and https://github.com/OSOceanAcoustics/echopype/pull/806, I've thought about tests. There are no tests (I think) that check for the correct creation of these file provenance variables. More broadly, I don't think we do much testing on provenance values, and that's probably a tricky area. Rather than trying to create very narrow tests in this PR, I think we should have a broader discussion. Maybe that could be folded into provenance work in 0.6.4.

leewujung, 2023-05-25:
This is likely related to #1043 and #1042.

emiliom, 2023-05-25:
See also the new issue #1050