echostack-org/echopype #839: Incorrect behavior of `check_zarr_path` for `zarr_path=None`

b-reyes, 2022-10-10:
In `check_zarr_path` we do checks that make sure the provided `zarr_path` is a string and has the appropriate suffix. However, these checks will fail if `zarr_path=None`. We need to make changes in the following if statements:

```
# check that zarr_path is a string
if not isinstance(zarr_path, str):
    raise TypeError(f"zarr_path must be of type {str}")

# check that the appropriate suffix was provided
if not zarr_path.strip("/").endswith(".zarr"):
    raise ValueError("The provided zarr_path input must have '.zarr' suffix!")
```

leewujung, 2022-10-10:
Yes, also I think it'll be nice to accept `pathlib.Path` on local machines. I had to use `str()` to use it.

b-reyes, 2022-10-10:
> Yes, also I think it'll be nice to accept pathlib.Path on local machines. I had to use str() to use it.

As we discussed, we will not accept `pathlib.Path` right now. I have created issue #841, which reminds us to implement this in the future.

b-reyes, 2022-10-11:
PR #840 addressed this issue, closing.

closed 2022-10-11 by b-reyes