echostack-org/echopype #846: Error with EK60 .raw file

SinanGISP, 2022-10-12:
I receive this error:
    ch_id = datagram["channel_id"] if "channel_id" in datagram else datagram["channel"]
KeyError: 'channel'

when i ry to use open_raw. any idea?

leewujung, 2022-10-12:
@SinanGISP : Can you post the code you used, the error messages, and a sample file? The one line you posted does not give enough information for understanding the problem you have.

SinanGISP, 2022-10-12:
thank you for your reply.
The code I used:
`import echopype as pe
from echopype import open_raw
file = "file_name.raw" # sample file https://drive.google.com/drive/folders/1tv1tM-qoCx2G9LR5h-UjG6516ACCwU2T?usp=sharing
ed = open_raw(file, sonar_model='EK60')'

leewujung, 2022-10-13:
@SinanGISP : It is important that you provide the requested information. Here I was able to test the file you provided to identify the cause of the problem, but it would have been much more efficient if you copy-pasted the error messages in the first place. We often are not able to prioritize solving problems specific individual files have. 

@b-reyes : I looked into this file and the error actually happens in the set_groups stage and has nothing to do with the parser. It happens when `_append_channel_ping_data` is called under `ParseEK.rectangularize_data` as you can see below. Since this involves the `zarr_datagrams` I think it is better that you look into this problem.

Error messages:
```python
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Input In [2], in <cell line: 1>()
----> 1 echodata = open_raw(raw_file="./test_files/EX2201_EK60-D20220226-T064314.raw", sonar_model="EK60")

File ~/code_git/echopype/echopype/convert/api.py:468, in open_raw(raw_file, sonar_model, xml_path, convert_params, storage_options, offload_to_zarr, max_zarr_mb)
    464     p2z = Parsed2Zarr(parser)
    465     if (sonar_model in ["EK60", "ES70", "EK80", "ES80", "EA640"]) and (
    466         "ALL" in parser.data_type
    467     ):
--> 468         parser.rectangularize_data()
    470 setgrouper = SONAR_MODELS[sonar_model]["set_groups"](
    471     parser,
    472     input_file=file_chk,
   (...)
    477     parsed2zarr_obj=p2z,
    478 )
    480 # Setup tree dictionary

File ~/code_git/echopype/echopype/convert/parse_base.py:77, in ParseEK.rectangularize_data(self)
     75 # append zarr datagrams to channel ping data
     76 for dgram in self.zarr_datagrams:
---> 77     self._append_channel_ping_data(dgram, zarr_vars=False)
     79 # Rectangularize all data and convert to numpy array indexed by channel
     80 for data_type in ["power", "angle", "complex"]:
     81     # Receive data

File ~/code_git/echopype/echopype/convert/parse_base.py:400, in ParseEK._append_channel_ping_data(self, datagram, rx, zarr_vars)
    384 """
    385 Append ping by ping data.
    386 
   (...)
    394     whether one should account for zarr vars
    395 """
    397 # TODO: do a thorough check with the convention and processing
    398 # unsaved = ['channel', 'channel_id', 'low_date', 'high_date', # 'offset', 'frequency' ,
    399 #            'transmit_mode', 'spare0', 'bytes_read', 'type'] #, 'n_complex']
--> 400 ch_id = datagram["channel_id"] if "channel_id" in datagram else datagram["channel"]
    402 # append zarr variables, if they exist
    403 if zarr_vars and rx:

KeyError: 'channel'

```

SinanGISP, 2022-10-14:
I am sorry for my late response. below is the full error message:

Traceback (most recent call last):
  File "~\EK60.py", line 5, in <module>
    ed = open_raw(file, sonar_model='EK60')
  File "~\AppData\Local\ESRI\conda\envs\arcdev\lib\site-packages\echopype\convert\api.py", line 488, in open_raw
    parser.rectangularize_data()
  File "~\AppData\Local\ESRI\conda\envs\arcdev\lib\site-packages\echopype\convert\parse_base.py", line 77, in rectangularize_data
    self._append_channel_ping_data(dgram, zarr_vars=False)
  File "~\AppData\Local\ESRI\conda\envs\arcdev\lib\site-packages\echopype\convert\parse_base.py", line 398, in _append_channel_ping_data
    ch_id = datagram["channel_id"] if "channel_id" in datagram else datagram["channel"]
KeyError: 'channel'

hope this helps.

leewujung, 2022-10-24:
Hi @SinanGISP : just realized that we haven't circled back to you on this. This issue your file made apparent was addressed in the most recent release v0.6.3. Please give it a try and let us know if you continue to see problems. Thanks again for reporting this issue!

closed 2022-10-24 by leewujung

SinanGISP, 2022-10-24:
Hi @leewujung: still having the same id-channel error with echopype v0.6.3
Traceback (most recent call last):
  File "~\Local\ESRI\conda\envs\arcdev\lib\site-packages\IPython\core\interactiveshell.py", line 3437, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-10-0de96b139dc5>", line 1, in <module>
    ed = open_raw(file, sonar_model='EK60')
  File "~\Local\ESRI\conda\envs\arcdev\lib\site-packages\echopype\convert\api.py", line 447, in open_raw
    parser.rectangularize_data()
  File "~\Local\ESRI\conda\envs\arcdev\lib\site-packages\echopype\convert\parse_base.py", line 77, in rectangularize_data
    self._append_channel_ping_data(dgram, zarr_vars=False)
  File "~\Local\ESRI\conda\envs\arcdev\lib\site-packages\echopype\convert\parse_base.py", line 411, in _append_channel_ping_data
    ch_id = datagram["channel_id"] if "channel_id" in datagram else datagram["channel"]
KeyError: 'channel'

b-reyes, 2022-10-24:
Hello @SinanGISP. While I was looking into this error I found that the data within the file you provide above is not consistent with EK60 data. From my review, it appears that it is EK80 data. With echopype v0.6.3, could you instead do the following:
```
ed = open_raw(file, sonar_model='EK80')
```

If you are still having issues, please let us know.

leewujung, 2022-10-24:
Oh right, thanks @b-reyes , I completely forgot to mention this important detail!