echostack-org/echopype #857: Remove remnant unused "ALL" param in parser object

leewujung, 2022-10-15:
We decided (and implemented) a few releases ago that we will NOT allow selective parsing of only a subset of EK60/EK80 datagrams. However the old selector `self.data_Type = "ALL"` is still present in the code and causing confusion. Let's remove it.

emiliom, 2023-03-16:
Is this as simple as removing these two blocks?

https://github.com/OSOceanAcoustics/echopype/blob/f8082cac32bfbee2b21c62d266507491d221182d/echopype/convert/parse_ek80.py#L15-L16

https://github.com/OSOceanAcoustics/echopype/blob/f8082cac32bfbee2b21c62d266507491d221182d/echopype/convert/parse_ek60.py#L12-L13

leewujung, 2023-03-17:
I am not sure, but it likely involves checking some functions that used ALL as a param. I think those were meant to try to selectively parse only some datagram types. Not sure if the mechanisms were entirely removed.

closed 2023-11-08 by praneethratna