echostack-org/echopype #872: Memory consumption and accumulation, and performance issues, in v0.6.3

emiliom, 2022-10-19:
When re-running and updating the [echopype-examples "hake" notebook](https://osoceanacoustics.github.io/echopype-examples/ms_PacificHake_EK60_cruisetracks.html) with echopype 0.6.3, I found that in the main data processing for-loop (in [section 2.3.1](https://osoceanacoustics.github.io/echopype-examples/ms_PacificHake_EK60_cruisetracks.html#echopype-processing)), the processing sequence of `open_raw`, `compute_Sv` and `compute_MVBS` is consuming more memory than it did previously with echopype 0.6.0. It is also significantly slower.

Initial assessments suggest the slowdown involves mainly `compute_MVBS`, which may be related to #845. We haven't done line-by-line profiling or analogous assessments to identify which step is leading to the increased memory usage relative to v. 0.6.0. 

Also, that for loop which repeats the same set of steps on a different raw file, overwriting previous variables, leads to the accumulation of memory that's not released at the end of each loop cycle. This may be a broader Python memory management limitation, but it's worth looking into, too.

leewujung, 2022-11-24:
The compute_MVBS efficiency issues are resolved by #878 .

leewujung, 2023-03-09:
@lsetiawan : I don't think we necessarily have to do this in v0.7.0 (ie before end of March), but I think it'll be good to do some initial investigation to know what problems we may be looking at.

emiliom, 2023-03-14:
I've rerun the hake notebook ([ms_PacificHake_EK60_cruisetracks.ipynb](https://osoceanacoustics.github.io/echopype-examples/ms_PacificHake_EK60_cruisetracks.html) with 0.6.4 on the same laptop I ran it in October with 0.6.3. The main data processing for-loop now took 8 minutes, compared to 13 minutes with 0.6.3. (I never wrote down what it took when running the notebook with 0.6.0 on the same laptop)

lsetiawan, 2023-03-14:
I've started a private repository for benchmarking and profiling at https://github.com/OSOceanAcoustics/benchmarking. You can find the code for the profiling below in the echopype directory. If you have mamba in your base environment and GNU make, simply update the `CONDA_ENV_DIR` variable in the `Makefile` and then run `make .PHONY` to run all of the process to perform this profiling.

## Memory Profiling

I've done memory usage profiling for v0.6.0, v0.6.3, and v0.6.4. It seems like the v0.6.4 is the fastest but takes the most memory. See the image below, the different versions is listed from top to bottom or colors (BLK - 0.6.0, BLU - 0.6.3, RED - 0.6.4).

**NOTE: I didn't run this on all 170 files, but rather only a subset (10 files)**

[image]

lsetiawan, 2023-11-06:
## Current `dev` branch

Similar benchmarking as above with the current echopype `dev` branch with the new flox based `compute_MVBS` can be seen in the following graph.

[image]

Looking at the resulting axis ranges, we can see that the current *memory usage* caps at a similar level to echopype `v0.6.0 - Black Line`. The *compute time* is closely similar to `v0.6.4`, though it is a little bit slower. 

Looking at this, I think the current branch get the best of both version: **lower memory footage** AND **faster computation**

closed 2023-11-06 by lsetiawan

leewujung, 2023-11-07:
Yes, this is exciting! We should try this on a much larger cluster to see what happens!

emiliom, 2023-11-07:
Wonderful!!

PS. This issue is currently tagged as Milestone 0.8.1. I'll change it to 0.8.2, now that  it's been fully addressed.