echostack-org/echopype #883: Using tempfile on Windows seems problematic

lsetiawan, 2022-11-19:
## Overview

As seen on PR #879 `test_utils_log.py` tests fails when using `TemporaryDirectory`. The error occurs when the built-in `tempfile` package tries to clean up the directory, see call stack:

```python
 C:\Users\runneradmin\micromamba-root\envs\echopype\lib\tempfile.py:834: in cleanup
    self._rmtree(self.name)
C:\Users\runneradmin\micromamba-root\envs\echopype\lib\tempfile.py:816: in _rmtree
    _shutil.rmtree(name, onerror=onerror)
C:\Users\runneradmin\micromamba-root\envs\echopype\lib\shutil.py:759: in rmtree
    return _rmtree_unsafe(path, onerror)
C:\Users\runneradmin\micromamba-root\envs\echopype\lib\shutil.py:629: in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
C:\Users\runneradmin\micromamba-root\envs\echopype\lib\tempfile.py:808: in onerror
    cls._rmtree(path)
C:\Users\runneradmin\micromamba-root\envs\echopype\lib\tempfile.py:816: in _rmtree
    _shutil.rmtree(name, onerror=onerror)
C:\Users\runneradmin\micromamba-root\envs\echopype\lib\shutil.py:759: in rmtree
    return _rmtree_unsafe(path, onerror)
C:\Users\runneradmin\micromamba-root\envs\echopype\lib\shutil.py:610: in _rmtree_unsafe
    onerror(os.scandir, path, sys.exc_info())
```

the errors are all coming from builtin functions.

With this in mind, it might be best to create a temporary directory ourselves, maybe in the user directory.

leewujung, 2022-11-23:
@lsetiawan : this is just for tests, right? If so, we can address this as part of the test overhaul, which perhaps we use  the approach you said to create a temp directory ourselves.

LOCEANlloydizard, 2025-10-20:
Hi @lsetiawan, I also had this problem.
I’ve proposed a fix in PR #1563 that addresses it, along with the Windows CI integration.
Cheers!