echostack-org/echopype #886: Require destination path when using `open_raw(..., offload_to_zarr=True)`

leewujung, 2022-11-24:
Currently when using `open_raw(..., offload_to_zarr=True)` a temporary zarr store is created in the "local" path to handle potential large expansion of memory for .raw files with irregular shape across pings or across channels (e.g., a common case is when only one channel is splitbeam and produces athwartship/alongship data). This is potentially problematic in a cloud environment due to the required disk space in the attached volume. We should modify the code so that users are required to provide the final zarr store location when using `offload_to_zarr=True` to avoid this problem. This need surfaced again while reviewing code for #879.

## Tasks
- [ ] instead of writing the expanded data into a temporary zarr store, "short-circuit" the implementation so that the expanded data are written directly into the destination path where the final file would live, such as an s3 bucket
- [ ] reuse `validate_output_path` when handling the destination path

lsetiawan, 2023-01-23:
Action item: @b-reyes and @lsetiawan need to connect and discuss this for transition.

leewujung, 2023-03-07:
@lsetiawan @valentina-s : could you summarize your conversation here so we are clear whether this should be closed or updated with whatever changes that you guys arrived on in the last discussion? Thanks!

lsetiawan, 2023-03-07:
Based on my notes we've decided on the following, which makes this issue stale:
- `offload_to_zarr` should be renamed to `use_swap` as it is a more familiar term and users don't really need to know what data format we're using under the hood
- There will NOT be a required destination path. `use_swap` will use the local drive and the "spilled" memory data files will be stored in echopype's temp folder.
- For data that are way too large and exceed the capacity of local drive, an error will be thrown

At some point we'll have to figure out how to distribute this capability!

closed 2023-03-07 by lsetiawan