echostack-org/echopype #889: Create and move functions to `unify` subpackage

leewujung, 2022-11-28:
To manage functions that all could be called as "processing" or "preprocessing" type, in #817 we laid out some subpackage organizations. This issue addresses one of them, to create a new subpackage called `unify` and move a subset of the functions currently under `preprocess` under it:
- `unify`
  - `compute_MVBS` (existing) -- the efficiency was enhanced in #878
  - `compute_MVBS_index_binning` (existing) -- we know more work may be needed for this, but we won't change the implementation at the moment.
  - `regrid_Sv` (to be implemented) -- See https://github.com/OSOceanAcoustics/echopype/issues/726

leewujung, 2022-11-29:
I don't think "unify" and "consolidate" are exactly synonyms, but if there are other names that are better we can use those. 

What I think the PR addressing this issue should achieve is to move those functions into a separate subpackage that is not called "preprocess." I think we have until v0.6.4 is released to finalize what subpackage name we want. 😛

emiliom, 2023-03-16:
Currently there is a third user function under `preprocess`: `remove_noise`. Our ultimate goal is to move (reorganize) it to a new subpackage, `filter` (per https://github.com/OSOceanAcoustics/echopype/pull/817#issuecomment-1294332886). I think we should try to move it at the same time as we move `compute_MVBS` and `compute_MVBS_index_binning`, so we can retire `preprocess` all at once.

BTW, as we move these functions, we should look into maintaining working stubs under `preprocess` that simply point to the new location of the functions and emit a warning that function calls using `preprocess` will be retired in 0.7.1. eg, a request for `preprocess.compute_MVBS` will result in a request to `unify.compute_MVBS` plus a warning.

leewujung, 2023-03-16:
> I think we should try to move it at the same time as we move `compute_MVBS` and `compute_MVBS_index_binning`, so we can retire `preprocess` all at once.

I like this idea, this way we won't be moving things around that much afterwards.

Also agreed with deprecation warning and remove everything in 0.7.1.

emiliom, 2023-03-17:
We've settled on `commongrid` instead of `unify`. PR #993 addresses the reorganization of functions to both `commongrid` and `filter`. BUT, we now have an open question about `filter`, because it's a Python reserved word :weary:

emiliom, 2023-03-17:
Not as good as `filter`, but I think it works.

emiliom, 2023-03-21:
Done in #993. Closing.

closed 2023-03-21 by emiliom