echostack-org/echopype #897: Utilize echopype working directory

lsetiawan, 2022-12-08:
## Overview

After the acceptance of the idea for an echopype working directory, the team have decided to do the following after PR #896 is merged:

- move all the temp directories and temp zarr files into this hidden directory, instead of the current `temp_echopype_output` directory
- move all the testing framework and files into this hidden directory

_Originally posted by @leewujung in https://github.com/OSOceanAcoustics/echopype/issues/896#issuecomment-1343043697_

leewujung, 2023-02-24:
A note that I was just looking at what `combine_echodata` does -- seems that if no destination path is given, the combined file will get saved at `temp_echopype_output/combined_echodatas.zarr` (see [here](https://github.com/OSOceanAcoustics/echopype/blob/e73fa14e28d69e1f254c40859eb23d683d76bec5/echopype/echodata/combine.py#L61)). 

I have two complaints:
1 . echodataS doesn't make sense to me -- how about just `combined _echodata.zarr`?
2. currently in [`io.validate_output_path`](https://github.com/OSOceanAcoustics/echopype/blob/e73fa14e28d69e1f254c40859eb23d683d76bec5/echopype/utils/io.py#L150), the way docstring is written it is not clear that whatever in `source_file` is ignored if `save_path` is a full path (including filename) and not just a path to a folder: https://github.com/OSOceanAcoustics/echopype/blob/e73fa14e28d69e1f254c40859eb23d683d76bec5/echopype/utils/io.py#L214-L230

@lsetiawan : what do you think about making these changes as you work on this issue to use the hidden default directory? For 2. it could just a docstring change for clarification, but it feels a little bizarre to have `source_file` and `save_path` partially overlap only  sometimes. Thoughts?  (I am fine with keeping it as is to not disturb anything major, but thought we should discuss regardless.)

leewujung, 2023-03-11:
This is addressed in #954.

closed 2023-03-11 by leewujung