echostack-org/echopype #941: Add rechunking option for the echodata object

valentina-s, 2023-02-13:
The recent PR #939 provides the option to have a default rechunking in `to_zarr`. In some cases, though, maybe the user wants to rechunk all data variables in the object in a very specific format good for their follow up analysis. It would be nice to have something like `echodata.(re)chunk(...)`

A few considerations brought up by @lsetiawan and @b-reyes :
* checking if the objects are already chunked
* chunk only those that are not chunked, or have some default option to overwrite existing chunking
* option to specify which variables to (re)chunk

leewujung, 2024-06-25:
@ctuguinay: This seems to be what you have implemented in #1331 under `utils.coding.py::chunk_echodata`?

ctuguinay, 2024-06-25:
Yeah it fits Valentina's request, but I'm wondering if this should be a class method? Like `ed.chunk({'ping_time':1000})`

I also did not see this issue before I started writing that PR 😆

leewujung, 2024-06-25:
I like the idea of making it a class method, and it would be pretty clean to have `ed.chunk({...})`.

leewujung, 2024-07-05:
This is added in #1331.

closed 2024-07-05 by leewujung