echostack-org/echopype #943: Calibration issues with environmental parameters (EK80)

liliaguillet, 2023-02-24:
### General description of problem

I am new working on acoustic data and I am trying to use echopype to calibrate my data from an EK80 sounder (ES200-7CDK-Split). I had my environmental parameters as explaining in your examples but when I run the calibration, I got an error message about "coords" attribute (see below). 
I didn't find any similar requests before.(or I missed it) 
I don't know if it is something I am doing wrong or if it is a bug. Could you help me please ? Thanks a lot for your help

### Computing environment

- Echopype version: 0.6.3
- How echopype was installed: pip install echopype
- Operating system: Windows 11 pro

### Minimum example

The following code reproduces the errors I encountered:
```python

raw_echodata = ep.open_raw(path, sonar_model="EK80")
ds_Sv_raw = ep.calibrate.compute_Sv(
        raw_echodata,
        env_params = {
            'temperature' :20.338, # Temperature in celsius
            'salinity' :34.27, # Salinity from PPS78 
            'pressure' :10.1325 # Pressure in dBar
            },
        waveform_mode="BB",
        encode_mode="complex",
    )
```

### Error message printouts

Below is the error messages I received when running the above code:
```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\lguillet\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\echopype\calibrate\api.py", line 205, in compute_Sv
    return _compute_cal(cal_type="Sv", echodata=echodata, **kwargs)
  File "C:\Users\lguillet\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\echopype\calibrate\api.py", line 89, in _compute_cal
    cal_ds = cal_obj.compute_Sv(waveform_mode=waveform_mode, encode_mode=encode_mode)
  File "C:\Users\lguillet\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\echopype\calibrate\calibrate_ek.py", line 997, in compute_Sv
    return self._compute_cal(
  File "C:\Users\lguillet\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\echopype\calibrate\calibrate_ek.py", line 963, in _compute_cal
    ds_cal = self._cal_complex(cal_type=cal_type, waveform_mode=waveform_mode)
  File "C:\Users\lguillet\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\echopype\calibrate\calibrate_ek.py", line 756, in _cal_complex
    if "channel" in self.env_params[p].coords:
AttributeError: 'float' object has no attribute 'coords'
```

### Provide an example file

- Link to an example raw file: [Example error.zip](https://github.com/OSOceanAcoustics/echopype/files/10822500/Example.error.zip)

### Troubleshooting
 I looked through the script Calibrate EK where the error appears (line 756). And I didn't see any moment where coord where added to env param before the line where these coords are asked. From this I was wondering what are these coords required, should I implement them in env params dictonnary ?

leewujung, 2023-02-27:
Hi @liliaguillet : Thanks for reporting this! This is definitely a bug that happened while we updated other parts of the code, sorry about that. There is a calibration overhaul in the working and I can get the fix into this upcoming release, but that will be early next week the earliest. If you could like to tweak the code in the meaning, you could add a check to only go into the line you pointed if `self.env_params` is an `xr.DataArray`. This should bypass the errorer out check altogether since your env parameters are scalars. Something like (I have not tested this):
```python
if isinstance(self.env_params[p], xr.DataArray):
  if "channel" in self.env_params[p].coords:
      self.env_params[p] = self.env_params[p].sel(channel=self.chan_sel)

```

liliaguillet, 2023-02-28:
Hi !

Thank you for your fast answer ! I tried your correction and it is working.

leewujung, 2023-02-28:
Great! Let's leave this issue open so I could close it properly in a new PR.

leewujung, 2023-03-11:
See https://github.com/OSOceanAcoustics/echopype/issues/950#issuecomment-1464962891

leewujung, 2023-05-17:
This issue is addressed, so closing this now.

closed 2023-05-17 by leewujung