echostack-org/echopype #950: Calibration issues with "gain_correction" parameters (EK80)

liliaguillet, 2023-02-28:
### General description of problem
> Hello, I have a new issue when I try to calibrate my data.  I am trying to use echopype to calibrate my data from an EK80 sounder (ES200-7CDK-Split). Now with correction from #943 issue the new environmental parameters are implemented. Now I am adding the calibration parameters (equivalent_beam_angle and gain_correction). I don't get any error but when I am printing my calibrated data, gain_ correction = None. It is not the case for equivalent beam angle and if I try to change the value of equivalent beam angle, the printed value is modified too. So the modification of equivalent beam angle is working but no gain correction is applied. 

I don't know if it is something I am doing wrong or if it is a bug. Could you help me please ? Thanks a lot for your help
> 
### Computing environment
> - Echopype version: 0.6.3
> - How echopype was installed (e.g., conda or pip):pip install echopype
> - Operating system:Windows 11 pro
> 
### Minimum example

> The following code reproduces the issue I encountered:
> ```python
> 
> raw_echodata = ep.open_raw(path, sonar_model="EK80")
> ds_Sv_raw = ep.calibrate.compute_Sv(
>         raw_echodata,
>         env_params = {
>             'temperature' :20.338, # Temperature in celsius
>             'salinity' :34.27, # Salinity from PPS78 (R gsw_SP_from_C)
>             'pressure' :10.1325 # Pressure in dBar
>             },
>         cal_params = {
>           'gain_correction': 28.49, # Center value of gain (here 19th frequency: 202657)
>          'equivalent_beam_angle' : -21.0 # Easy to find ctrl+F in the calibration file
>            },
>         waveform_mode="BB",
>         encode_mode="complex",
>     )
>     
> ```
> 
### Output: ds_Sv_raw
> ```python
> <xarray.Dataset>
> Dimensions:                (channel: 1, ping_time: 42, range_sample: 17149,
>                             filenames: 1, time3: 1)
> Coordinates:
>   * channel                (channel) <U30 'EKA 278996-07 ES200-7CDK-Split'
>   * ping_time              (ping_time) datetime64[ns] 2023-01-08T05:31:22.778...
>   * range_sample           (range_sample) int32 0 1 2 3 ... 17146 17147 17148
>   * filenames              (filenames) int32 0
>   * time3                  (time3) datetime64[ns] 2023-01-08T05:31:21.118000128
> Data variables: (12/13)
>     Sv                     (channel, ping_time, range_sample) float64 nan ......
>     echo_range             (channel, ping_time, range_sample) float64 0.0 ......
>     frequency_nominal      (channel) float64 2e+05
>     temperature            float64 20.34
>     salinity               float64 34.27
>     pressure               float64 10.13
>     ...                     ...
>     sound_absorption       (channel, ping_time) float64 0.08563 ... 0.08563
>     sa_correction          object None
>     gain_correction        object None
>     equivalent_beam_angle  float64 -21.0
>     source_filenames       (filenames) <U65 '..\\sailor_data/SLUAquaSailor202...
>     water_level            (time3) float64 nan
> Attributes:
>     processing_software_name:     echopype
>     processing_software_version:  0.6.3
>     processing_time:              2023-02-28T14:41:20Z
>     processing_function:          calibrate.compute_Sv
> ```
> 
### Example
> 
> - Link to an example raw file: [Example error.zip](https://github.com/OSOceanAcoustics/echopype/files/10851422/Example.error.zip)
> 
### Troubleshooting
> First, I though that maybe it was a printing issue of the gain correction in the output and the correction was applied during the calibration. To verify this theory I tried to directly modify the gain correction (calibrate_ek line 177) adding this line : self.cal_params["gain_correction"] = 28.49. But when I compare the output with this modification and without calibration from what I saw there is no difference. So there is no gain correction applied. I don't understand why. 
>

leewujung, 2023-02-28:
@liliaguillet : thanks for highlighting this. This is something tracked in an issue (#887) but we haven’t had time to address until now. It will be either this upcoming release or next release. One way to “short circuit” this is to substitute the desired cal parameters into the echodata object. If your data is CW then the place to put these would be:
- gain_correction: echodata[“Vendor_specific”] — note there is an equivalent variable in the Sonar/Beam_group1 also but the code is using the values from the Vendor-specific group (please double check this as you have done already, because I’m on my phone due to travel and couldn’t step through the code to be 100% certain right now)
- equivalent beam angle: echodata[“Sonar/Beam_group1”]

If you run into trouble substituting specific variables try SOME_VAR.data[SLICING_INDICES] since with xarray data array the slicing may be returning a view only.

Sorry this is such a patchy solution right now, but after the upcoming two releases the EK80 calibration code will be way more robust, and include capability to use Echoview generated ECS files directly. 

If your data is broadband, we’ll have to take it with the upcoming release 😕.

liliaguillet, 2023-03-01:
Thanks a lot for your help !! 
Unfortunately my data is broadband. I will wait until the next version ! 
Do you know when the new version will be released ? :)

leewujung, 2023-03-01:
Aiming for next week, lots of things happening! 🤞

leewujung, 2023-03-11:
Hi @liliaguillet : could you please give the current `dev` a try? you can pip install that by:
```
pip install git+git://github.com/OSOceanAcoustics/echopype.git@dev
```
This also include fixes to #943.
We have everything ready to go and plan to release on Monday. :)

leewujung, 2023-05-17:
This issue is addressed, so closing this now.

closed 2023-05-17 by leewujung