echostack-org/echopype #957: Update the behavior of `offload_to_zarr` to be more user friendly

lsetiawan, 2023-03-07:
## Overview              

Based on in-person conversation notes we've decided on the following for `offload_to_zarr` functionality:
- `offload_to_zarr` should be renamed to `use_swap` as it is a more familiar term and users don't really need to know what data format we're using under the hood
- There will NOT be a required destination path. `use_swap` will use the local drive and the "spilled" memory data files will be stored in echopype's temp folder.
- For data that are way too large and exceed the capacity of local drive, an error will be thrown

At some point we'll have to figure out how to distribute this capability!

_Originally posted by @lsetiawan in https://github.com/OSOceanAcoustics/echopype/issues/886#issuecomment-1458498205_

leewujung, 2023-03-08:
Great, thanks for detailing these. 

> At some point we'll have to figure out how to distribute this capability!

Yes, this would take some benchmarking probably to decide if we should consider distributing this, or solving on the data organization side, ie figuring out what are the cases where we may not want to pad NaNs -- I've been thinking about this for the setups that ping sequentially across channels.

lsetiawan, 2023-11-06:
The actual issue that tracks this work is located in Issue #1179

closed 2023-11-06 by lsetiawan