echostack-org/echopype #963: Update Windows CI so it can also work to run the full test suite

lsetiawan, 2023-03-08:
## Overview

Currently the Windows CI only runs the `utils` module and NOT the full test suite for echopype (https://github.com/OSOceanAcoustics/echopype/actions/workflows/windows-utils.yaml). This needs to be updated so that the whole test suite will run for Windows (https://github.com/OSOceanAcoustics/echopype/actions/workflows/windows.yaml). The backbone is there, but it's currently not activated due to Docker limitations in working with Windows.

To make this work, we can potentially do the following:
- Manually setup minio straight into the CI server: https://github.com/minio/minio#microsoft-windows
- Manually setup apache http server into the CI server: https://httpd.apache.org/docs/2.4/platform/windows.html
- Perform rclone at the beginning of the CI run with the `gdrive-rclone` action that we have: https://github.com/OSOceanAcoustics/echopype/tree/main/.github/actions/gdrive-rclone

LOCEANlloydizard, 2025-08-07:
Hi all,

As I'm developing on Windows, I encountered errors locally when running the tests (same as #883). I wanted to compare with Windows runners, but as mentioned in this issue there's Docker limitations in working with Windows (from what I've read, it will stay like this).

@leewujung mentioned GitHub assets as an alternative, so I explored that and managed to get a Windows runner working on my fork, pulling data from a GitHub release instead of Docker. For the integration, as I don’t have write access to the main echopype repo, a maintener needs to create and load the test data in the release (I've written what worked for me in my fork below).

After running the test, and after a quick look at the test results, I’m seeing the same errors as my local ones and as well as #883, mostly related to how temporary files are handled on Windows. I’ll dig into that soon, I think it's just a matter of how the test close the files?

[image]

################ Workflow to import data in github assets to have Windows CI runner for echopype

**1. Extract the test data from docker** (if you don't have it locally)

We first pull the latest Docker images and deply test data services locally using the helper script:

    `python .ci_helpers/docker/setup-services.py --deploy`

This copies a large set of test data folders into the local echopype/test_data directory.

**2. Zip each test data folder separately**

The _**separately**_ is important to overcome the GitHub Release asset import size limit (~2GB), so we zip each subfolder individually instead of the entire test data at once.

Example PowerShell code to zip each folder:

```
    $folders = Get-ChildItem -Directory -Path "echopype/test_data"
    foreach ($folder in $folders) {
      $zipFile = "echopype/${folder.Name}.zip"
      Compress-Archive -Path $folder.FullName -DestinationPath $zipFile
    }
```

This creates multiple zip files like ek80.zip, ad2cp.zip, etc.

**3. Create a GitHub Release with the zip files**

We created a GitHub release (e.g. v0.1.0) to host these zipped test datasets as assets.
Using GitHub CLI (gh) from your fork repository:

```
    gh release create v0.1.0 --title "Test Data" --notes "Windows CI test data"
    gh release upload v0.1.0 *.zip
```
(Note: The gh CLI needs to be configured with the correct repo context and authentication.)
Once this step is done, the Test Data is visible as a release [here](https://github.com/OSOceanAcoustics/echopype/releases)

**4. Download and extract test data in Windows Runner via GitHub Release Assets**

In the GitHub Actions workflow for Windows, test data is downloaded by looping over all release asset zip files, for example using PowerShell:

```
    - name: Download test data from GitHub Release (Windows only)
      shell: pwsh
      run: |
        $urls = @(
          "ad2cp", "azfp", "azfp6", "ea640", "ecs", "ek60", "ek60_calibrate_chunks",
          "ek60_missing_channel_power", "ek80", "ek80_bb_complex_multiplex", "ek80_bb_with_calibration",
          "ek80_duplicate_ping_times", "ek80_ext", "ek80_invalid_env_datagrams",
          "ek80_missing_sound_velocity_profile", "ek80_new", "ek80_sequence",
          "es60", "es70", "es80", "legacy_datatree"
        )
        foreach ($name in $urls) {
          $zip = "${name}.zip"
          Invoke-WebRequest -Uri "PATH_OF_RELEASE/$zip" -OutFile $zip
          Expand-Archive -Path $zip -DestinationPath ./echopype/test_data -Force
        }
```

This last step comes jointly with a rewriting of the pr.yaml, which requires slight adjustments (i've adapted the pr.yaml [here](https://github.com/LOCEANlloydizard/echopype/blob/main/.github/workflows/pr.yaml) ; we could use the same with minor adaptations).

Overall, this approach allows us to:

- Use the same test suite on Windows and Linux runners.
- Avoid Docker-related issues on Windows CI by using pre-packaged test data assets.
- Keep test data manageable and under GitHub asset size limits.

LOCEANlloydizard, 2025-08-08:
This is linked with PR #1032.