echostack-org/echopype #973: Priority documentation updates

emiliom, 2023-03-10:
The documentation is lagging behind important developments since probably 0.6.2 or so. Let's create a list of priority gaps to address.

leewujung, 2023-03-11:
Add examples on where files are saved to docs: see https://github.com/OSOceanAcoustics/echopype/pull/954#discussion_r1132905334 on examples already laid out in docstrings.

leewujung, 2023-03-13:
@emiliom : I decided to just edit the issue description instead of adding comments to this, so that it's easy to keep track of what's done or not.

leewujung, 2023-03-13:
@emiliom will do the first cut on what to add for v0.7.0

leewujung, 2025-05-01:
Most of the items in the list were added in summer 2024. I'll close this now and when we have a chance we'll add a new issue for adding other undocumented functions.

closed 2025-05-01 by leewujung