echostack-org/echopype #978: beam dimension in EK80 calibrated data

emiliom, 2023-03-13:
This is a a subset of #684 that would be very helpful to address in milestone 0.7.0.

@leewujung : can you flesh out the task, edit the issue title, etc?

leewujung, 2023-03-17:
@emiliom : I've updated the description above. Let me know if you have questions.

emiliom, 2023-05-12:
> EK80 has >1 length in the `beam` dimension, but these beams are actually within the same transducer (channel).

> Note that these stored values in the Beam_groupX or Vendor-specific group (if exist, some may not) are transducer-specific and therefore do not vary with `beam` or `ping_time`.

@leewujung I'm trying to see if I understand. You're saying there are multiple beams within each transducer (channel), but the values of the variables you've listed are constant across beam (and ping_time), *within* each channel, right?

Do other EK80 variables that currently have a `beam` dimension vary across the `beam` dimension? The main such variables are `backscatter_r` and `backscatter_i`, but these also occur:

```
beam_direction_x
beam_direction_y
beam_direction_z
angle_alongship
angle_athwartship
frequency_start
frequency_end
```

emiliom, 2023-05-23:
For reference: the addition of the `beam` dimension was originally discussed in #520

leewujung, 2023-07-15:
> @leewujung I'm trying to see if I understand. You're saying there are multiple beams within each transducer (channel), but the values of the variables you've listed are constant across beam (and ping_time), _within_ each channel, right?

Yes, they are constant across beam and ping_time, _within_ each channel.

> Do other EK80 variables that currently have a `beam` dimension vary across the `beam` dimension? The main such variables are `backscatter_r` and `backscatter_i`, but these also occur:
>
> ``` [...]

For EK80 complex samples (both `backscatter_r` and `backscatter_i` exist):
- the `beam` dimension is necessary and valid -- they are the transducer sectors/quadrants or the "subbeam" dimension in ver.2 definition.
- `angle_alongship` and `angle_athwartship` do not exist in the raw data -- they need to be computed using `add_splitbeam_angle` (right now we have that to add to Sv datasets)
- `frequency_start` and `frequency_end` can vary across ping_time, but there is no beam dimension
- `beam_direction_x/y/z` does not have beam dimension and do not vary across ping_time (for the rigid installation for most echosounders anyway)

For EK80 power/angle samples (only `backscatter_r` exists)
- the data is equivalent with EK60 format, so the `beam` dimension is redundant
- `angle_alongship` and `angle_athwartship` exist in the raw data with the same dimensions as `backscatter_r`
- `frequency_start` and `frequency_end` do not exist -- since the data is narrowband
- `beam_direction_x/y/z` does not have beam dimension and do not vary across ping_time (for the rigid installation for most echosounders anyway)

leewujung, 2023-07-24:
Addressed in #1056.