echostack-org/echopype #1009: Error during interpolation in `consolidate.add_latlon` with an edge case

emiliom, 2023-03-25:
(Following up from https://github.com/OSOceanAcoustics/echopype/issues/769#issuecomment-1482171650) In PR #1000 I added several new `consolidate.add_latlon` tests to `test_consolidate_integration.py`, including two new test datasets (including one AZFP). One thing I did *not* add was an EK80 test file. While running `consolidate.add_latlon` on two EK80 test files just now, I ran into the same error for both files during the interpolation step (`return position_var.interp(time1=ds["ping_time"])`):
```
InvalidIndexError: Reindexing only valid with uniquely valued Index objects
```

The two files are `ek80_bb_with_calibration/2018115-D20181213-T094600.raw` and `ek80/Summer2018--D20180905-T033113.raw`. Examining the data a bit, I found two factors in common among those two files, other than being EK80:
- Their time-varying lat-lon locations are effectively at a single point, +/- very small floating-point variations.
- They span a tiny time range, of a couple of seconds.

The error is actually emitted by Pandas. In searching for this error message, I see some references to problems with datetimes. I wasted a lot of time thinking the problem might be with the interpolation of near-constant values, but I now think it's somehow related to the tiny datetime range. I don't think it has anything to do with the files being from EK80. It'll take more effort to track this down.

leewujung, 2023-03-25:
Well, I am not sure where the test files came from, but the small time (and hence lat/lon) range is likely inherent from small files. The per-ping data size is often larger, so the number of pings is smaller for a file that is just a couple MB.

leewujung, 2025-09-21:
Seems like this can be closed now with #1294 closed.