echostack-org/echopype #1014: Add temporal and spatial info to `compute_NASC` outputs

leewujung, 2023-03-25:
Leftover from #1005: 

We should consider adding the mean time, latitude, and longitude over the NASC cells with a `distance` dimension to the `compute_NASC` output.

See discussions in https://github.com/OSOceanAcoustics/echopype/pull/1005#issuecomment-1483404829
>> short of adding the mean time, lon and lat variables (with distance dim) that I mentioned, if we wanted to do it now.
>
> I have mixed feeling about this, mainly two things:
> 
> - right now the distance and depth dimensions uses the starting bin edges and not the mean or mid point of the bins (see next point) as the values -- we should probably stick with one scheme
> - for the means: should it be true means (as in the mean of the values), or should it just be the middle point of the bins?

leewujung, 2023-10-14:
This can be closed once #1127 is merged.

emiliom, 2023-10-14:
Did you mean PR #1167? #1127 is an issue.

Also, add a processing level decorator to `compute_NASC` in PR #1167, once compute_NASC is ready? `@add_processing_level("L4*")`

leewujung, 2023-11-11:
We can close this now!

closed 2023-11-11 by leewujung