echostack-org/echopype #1026: [File conversion] Time with <M8 short string notation fail to open on EC2 Amazon Linux 2023

oftfrfbf, 2023-04-06:
### General description of problem
Problem opening EK60 files on Arm architecture. Parser with ping_time dtype in short string notation of '<M8[ms]' fails assertion when compared to 'datetime64[ns]'.

### Computing environment
- Echopype version: 0.7.0
- How echopype was installed:
```bash
python3 -m pip --version
pip 21.3.1 from /usr/lib/python3.9/site-packages/pip (python 3.9)
```
- Operating system: Amazon Linux 2023

### Minimum example
The following is code that works properly on my local notebook:
Environment:
```bash
# Apple M1 Pro, macOS 13.3 (22E252)
user$ uname -a
Darwin user 22.4.0 Darwin Kernel Version 22.4.0: Mon Mar  6 20:59:28 PST 2023; root:xnu-8796.101.5~3/RELEASE_ARM64_T6000 arm64
user$ python
Python 3.9.6 (default, Mar 10 2023, 20:16:38) 
[Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
```
Working Code:
```python
import boto3, os
import echopype as ep
from botocore.client import Config, UNSIGNED
s3 = boto3.client('s3', config=Config(signature_version=UNSIGNED))
print(f"echopype version: {ep.__version__}") # echopype version: 0.7.0
input_bucket = "noaa-wcsd-pds"
input_key = "data/raw/Henry_B._Bigelow/HB20ORT/EK60/D20200225-T163738.raw"
row_split = input_key.split(os.sep)
ship, cruise, sensor, filename = row_split[-4:]
s3.download_file(Bucket=input_bucket, Key=input_key, Filename=filename)
echodata = ep.open_raw(filename, sonar_model='EK60') ### CRASHING HERE ###
print(echodata)
#<EchoData: standardized raw data from Internal Memory>
#Top-level: contains metadata about the SONAR-netCDF4 file format.
#├── Environment: contains information relevant to acoustic propagation through water.
#├── Platform: contains information about the platform on which the sonar is installed.
#│   └── NMEA: contains information specific to the NMEA protocol.
#├── Provenance: contains metadata about how the SONAR-netCDF4 version of the data were obtained.
#├── Sonar: contains sonar system metadata and sonar beam groups.
#│   └── Beam_group1: contains backscatter power (uncalibrated) and other beam or channel-spec...
#└── Vendor_specific: contains vendor-specific information about the sonar and the data.
```

This is the code that crashes on an AWS EC2 Arm machine:
Environment:
```bash
# Amazon "t4g.medium" instance, Arm-based AWS Graviton2 processor, AMI ID: ami-01d9e06b75f9d69c4, Region: us-east-1
[ec2-user@ip-172-31-24-120 remote]$ uname -a
#Linux ip-172-31-24-120.ec2.internal 6.1.19-30.43.amzn2023.aarch64 #1 SMP Wed Mar 15 14:44:03 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
[ec2-user@ip-172-31-24-120 remote]$ python3
#Python 3.9.16 (main, Feb 23 2023, 00:00:00) 
#[GCC 11.3.1 20221121 (Red Hat 11.3.1-4)] on linux
```
Failing Code:
```python
>>> import boto3, os
>>> import echopype as ep
>>> from botocore.client import Config, UNSIGNED
>>> s3 = boto3.client('s3', config=Config(signature_version=UNSIGNED))
>>> print(f"echopype version: {ep.__version__}")
echopype version: 0.7.0
>>> input_bucket = "noaa-wcsd-pds"
>>> input_key = "data/raw/Henry_B._Bigelow/HB20ORT/EK60/D20200225-T163738.raw"
>>> row_split = input_key.split(os.sep)
>>> ship, cruise, sensor, filename = row_split[-4:]
>>> s3.download_file(Bucket=input_bucket, Key=input_key, Filename=filename)
>>> echodata = ep.open_raw(filename, sonar_model='EK60') ### CRASHES HERE!!! ###
```

### Error message printouts
_Required: Copy-paste the entire error messages you encounter here._
Below is the error messages I received when running the above code:
```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ec2-user/.local/lib/python3.9/site-packages/echopype/utils/prov.py", line 216, in inner
    dataobj = func(*args, **kwargs)
  File "/home/ec2-user/.local/lib/python3.9/site-packages/echopype/convert/api.py", line 479, in open_raw
    tree_dict["Environment"] = setgrouper.set_env()
  File "/home/ec2-user/.local/lib/python3.9/site-packages/echopype/convert/set_groups_ek60.py", line 210, in set_env
    return set_time_encodings(ds)
  File "/home/ec2-user/.local/lib/python3.9/site-packages/echopype/utils/coding.py", line 93, in set_time_encodings
    new_ds[var] = xr.apply_ufunc(
  File "/home/ec2-user/.local/lib/python3.9/site-packages/xarray/core/computation.py", line 1159, in apply_ufunc
    return apply_dataarray_vfunc(
  File "/home/ec2-user/.local/lib/python3.9/site-packages/xarray/core/computation.py", line 286, in apply_dataarray_vfunc
    result_var = func(*data_vars)
  File "/home/ec2-user/.local/lib/python3.9/site-packages/xarray/core/computation.py", line 727, in apply_variable_ufunc
    result_data = func(*input_data)
  File "/home/ec2-user/.local/lib/python3.9/site-packages/echopype/utils/coding.py", line 51, in _encode_dataarray
    encoded_data, _, _ = coding.times.encode_cf_datetime(
  File "/home/ec2-user/.local/lib/python3.9/site-packages/xarray/coding/times.py", line 607, in encode_cf_datetime
    assert dates.dtype == "datetime64[ns]"
AssertionError
```

### Provide an example file
- Link to an example raw file: File is provided in the example above, downloads from S3 bucket via boto3.
- Raw file sensor model: The file is EK60. Scanning the datagram confirms a header value of "CON0."

### Troubleshooting
I can remotely debugging the python session in EC2 and can see that the parser is reading the datetime64 dtype in a short string notation of '<M8[ms]'. The assertion in 'encode_cf_datetime' makes the comparison between the two and it fails. I am not entirely following why this succeeds on other arm64 architectures — the only significant difference I see between the two is the Clang vs GCC compilers. If you would like anymore details please let me know and I will follow up.

leewujung, 2023-04-07:
Hey @oftfrfbf : thanks for reporting this! Could you help look into this and propose a fix? We are unlikely to get to this soon due to too many other pending tasks. Thanks!

oftfrfbf, 2023-04-07:
Thanks for the reply @leewujung. I tracked the error back to here on line 152:
https://github.com/OSOceanAcoustics/echopype/blob/b48db38328ca41cdbfb87239b49381b72f928d87/echopype/convert/set_groups_ek60.py#L153 where the Xarray Dataset is created. On line 189 the two workflows diverge where the local version captures the dtype as '<M8[ns]' whereas Amazon Linux 2023 returns '<M8[ms]' (see below).
[image]
I will work on figuring out a solution. Right now my best guess is that the Xarray Dataset needs to have encodings specified for each array as an added argument.

emiliom, 2023-04-07:
Are the two environments identical, or is there a chance that a significant version difference in a dependency crept in?

I was looking through the [Pandas 2.0.0 release announcement](https://pandas.pydata.org/docs/whatsnew/v2.0.0.html) (came out April 3, Monday), and it struck me that there are many changes to the handling of datetime64 data. The contrast you highlight doesn't show a Pandas footprint, but Pandas could have an indirect effect somewhere. Any chance the EC2 environment has Pandas 2, if it was built after Monday, while the other has Pandas < 2?

oftfrfbf, 2023-04-07:
@emiliom good catch. I had different versions of pandas installed between the two systems.
MacOS
```bash
mac user:~ mac user$ python3 -m pip freeze | grep 'pandas'
pandas==1.5.1
```
Amazon Linux 2023
```bash
[ec2-user@ip-172-31-24-120 ~]$ python3 -m pip freeze | grep 'pandas'
pandas==2.0.0
```
Updating MacOS to pandas 2.0.0 is now causing the same assertion failure as the other system.
```python
>>> import boto3, os
>>> import echopype as ep
>>> from botocore.client import Config, UNSIGNED
>>> s3 = boto3.client('s3', config=Config(signature_version=UNSIGNED))
>>> print(f"echopype version: {ep.__version__}") # echopype version: 0.7.0
echopype version: 0.7.0
>>> input_bucket = "noaa-wcsd-pds"
>>> input_key = "data/raw/Henry_B._Bigelow/HB20ORT/EK60/D20200225-T163738.raw"
>>> row_split = input_key.split(os.sep)
>>> ship, cruise, sensor, filename = row_split[-4:]
>>> s3.download_file(Bucket=input_bucket, Key=input_key, Filename=filename)
>>> echodata = ep.open_raw(filename, sonar_model='EK60')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/echopype/utils/prov.py", line 216, in inner
    dataobj = func(*args, **kwargs)
  File "/Users/rudy/Library/Python/3.9/lib/python/site-packages/echopype/convert/api.py", line 479, in open_raw
    tree_dict["Environment"] = setgrouper.set_env()
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/echopype/convert/set_groups_ek60.py", line 210, in set_env
    return set_time_encodings(ds)
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/echopype/utils/coding.py", line 93, in set_time_encodings
    new_ds[var] = xr.apply_ufunc(
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/xarray/core/computation.py", line 1159, in apply_ufunc
    return apply_dataarray_vfunc(
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/xarray/core/computation.py", line 286, in apply_dataarray_vfunc
    result_var = func(*data_vars)
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/xarray/core/computation.py", line 727, in apply_variable_ufunc
    result_data = func(*input_data)
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/echopype/utils/coding.py", line 51, in _encode_dataarray
    encoded_data, _, _ = coding.times.encode_cf_datetime(
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/xarray/coding/times.py", line 607, in encode_cf_datetime
    assert dates.dtype == "datetime64[ns]"
AssertionError
>>> print(echodata)
```

emiliom, 2023-04-07:
Progress! I would stay away from Pandas 2.0.0 for now, when working with echopype. It's too new (brand new), and may cause a number of unexpected issues.

When you install echopype, try pinning Pandas to "<2". We should do the same in our end, going forward, to avoid surprises. If you use `conda` this should be straightforward. If you're using `pip`, I think you could also do something like `pip install echopype pandas<2`

oftfrfbf, 2023-04-08:
Thanks @emiliom, I am glad you were able to track the problem down. I will close this issue for my specific request. Hopefully the Pandas major release doesn't create too many new problems on the horizon.

closed 2023-04-08 by oftfrfbf

leewujung, 2023-04-10:
This does seem to be what's happening:
From [here](https://pandas.pydata.org/docs/dev/whatsnew/v2.0.0.html#disallow-astype-conversion-to-non-supported-datetime64-timedelta64-dtypes):
> In previous versions, converting a Series or DataFrame from datetime64[ns] to a different datetime64[X] dtype would return with datetime64[ns] dtype instead of the requested dtype. In pandas 2.0, support is added for “datetime64[s]”, “datetime64[ms]”, and “datetime64[us]” dtypes, so converting to those dtypes gives exactly the requested dtype