echostack-org/echopype #1027: Pin Pandas to <2?

emiliom, 2023-04-07:
[Pandas 2.0.0 was released a few days ago](https://pandas.pydata.org/docs/whatsnew/v2.0.0.html) (Monday 4/3/23). That's a **major** new version, and I think it's safe to expect breaking changes. Already, the issue reported yesterday in #1026 *may* be caused by Pandas 2.0.0.

Let's consider adding a `pandas<2` pin in [requirements.txt](https://github.com/OSOceanAcoustics/echopype/blob/main/requirements.txt) for now, to be safe.

leewujung, 2023-04-09:
I think pinning pandas at <2 is a good option for now, especially when other packages are catching with it, but we'll need to take care of the breaking changes at some point. I think toward the second half of the summer may be a good time when other more urgent tasks are resolved.

emiliom, 2023-04-20:
Implemented in PR #981. Closing

closed 2023-04-20 by emiliom