echostack-org/echopype #1040: [File conversion] SUSPECT_CAUSE

spacetimeengineer, 2023-05-05:
### General description of problem
Unable to convert .raw files due to an apparent date-time formatting issue. It is not clear if echopype is responsible or the .raw file is 'wrong' somehow. I posted a google drive link to the raw.

### Computing environment
_Required_t
- Echopype version: 0.7.0
- How echopype was installed (e.g., conda or pip): pip3
- Operating system: Ubuntu 22.04.2 LTS 64Bit

### Minimum example
_Required: A minimum code snippet that reproduces your problem._
The following code reproduces the errors I encountered:
```python
ed = ep.open_raw(raw_file = 'D20090405-T114914.raw', sonar_model='EK60') 
```

### Error message printouts
_Required: Copy-paste the entire error messages you encounter here._
Below is the error messages I received when running the above code:
```python
File "/home/mryan/Desktop/echocluster/examples/app.py", line 324, in run
    ed = ep.open_raw(raw_file = 'D20090405-T114914.raw', sonar_model='EK60')  
  File "/home/mryan/.local/lib/python3.10/site-packages/echopype/utils/prov.py", line 216, in inner
    dataobj = func(*args, **kwargs)
  File "/home/mryan/.local/lib/python3.10/site-packages/echopype/convert/api.py", line 479, in open_raw
    tree_dict["Environment"] = setgrouper.set_env()
  File "/home/mryan/.local/lib/python3.10/site-packages/echopype/convert/set_groups_ek60.py", line 210, in set_env
    return set_time_encodings(ds)
  File "/home/mryan/.local/lib/python3.10/site-packages/echopype/utils/coding.py", line 93, in set_time_encodings
    new_ds[var] = xr.apply_ufunc(
  File "/home/mryan/.local/lib/python3.10/site-packages/xarray/core/computation.py", line 1159, in apply_ufunc
    return apply_dataarray_vfunc(
  File "/home/mryan/.local/lib/python3.10/site-packages/xarray/core/computation.py", line 286, in apply_dataarray_vfunc
    result_var = func(*data_vars)
  File "/home/mryan/.local/lib/python3.10/site-packages/xarray/core/computation.py", line 727, in apply_variable_ufunc
    result_data = func(*input_data)
  File "/home/mryan/.local/lib/python3.10/site-packages/echopype/utils/coding.py", line 51, in _encode_dataarray
    encoded_data, _, _ = coding.times.encode_cf_datetime(
  File "/home/mryan/.local/lib/python3.10/site-packages/xarray/coding/times.py", line 607, in encode_cf_datetime
    assert dates.dtype == "datetime64[ns]"
AssertionError
```

### Provide an example file
_Required: An example file that allow reproducing the problem._
- Link to the raw file: https://drive.google.com/file/d/1bY82xCM_uo76QtXzZERKHrhIMr_MMrfQ/view?usp=sharing    Michael : Apologies, I realize this file is 50 MB. This file and a few other large files I have are the only files I have to work with and they were requested by another for me for process. Some context ; I am a machine-learning engineer who is using (and appreciates) echopype so that I can tae advantage of the xarray format for the application of machine learning operations. Also I work for Dr. Mike Jech in Woods Hole if that is relevent. I am build ML utilities for him.
- Raw file sensor model: 'EK60'

### Troubleshooting
I performed the necessary research and standard debugging practices to the point where I could ascertain that this particular problem lied outside the scope of my expertise. Also it seems nuanced. It is clearly a formatting issue but my understanding of the echopype model and raw data itself is non existent. I tried to fix it myself through a series of brute force methods but all were futile in the end and I learned nothing:)

leewujung, 2023-05-05:
Hey @spacetimeengineer -- thanks for reporting this! Could you give v0.7.1 a try? Because this is related to datetime, it is very possible it is because of the pandas 2.0 breaking changes. We updated the requirements to pandas <2 and unpinned xarray in v0.7.1 (xarray also pinned pandas< 2.0 for now). If this is the cause, hopefully we can unpin pandas at some point soon, especially when xarray unpins it.

spacetimeengineer, 2023-05-05:
> Oh, sure! I didn't realize there was a new version out. Thankyou

That fixed it! Thank-you.

leewujung, 2023-05-05:
Great!

closed 2023-05-05 by spacetimeengineer