echostack-org/echopype #1049: WBT mini EK80 file has env temp = np.ndarray

rolandproud, 2023-05-25:
For a WBT mini raw file (EK80) with 2 frequencies, the env temperature parameter is a numpy.ndarray of length 1. This causes an error in 

line 152, in sanitize_user_env_dict
raise ValueError(f"{p_name} has to be a scalar, list, or an xr.DataArray")

I added

elif isinstance(p_val, np.ndarray):
    out_dict[p_name] = p_val[0]

so that files will still process, but needs looking into to.

Cheers

emiliom, 2023-06-13:
Thanks for bringing this to our attention and sharing a solution!

leewujung, 2023-10-14:
@rolandproud: could you provide an example file for this so that we can look into this in detail? from the description I can't tell if this call was based on data stored in files or user-defined env_params dict.

leewujung, 2023-10-25:
@rolandproud : pinging again to see if this is still an issue. If so, could you provide an example file?

rolandproud, 2023-10-25:
Hi Wu-jung,

Please close the issue. I work part time now and I don’t have time to go back and recreate the problem. If I experience the same problem again using the new version I will send an example file.

Cheers,

Roland

closed 2023-11-05 by praneethratna