echostack-org/echopype #1053: use_swap= True,  TypeError: Object of type int32 is not JSON serializable

rolandproud, 2023-05-31:
I have tried use_swap = True on a number of different files (EK60/EK80/ES70) and everytime I get the same error:

ek = ep.open_raw(raw_file_name, sonar_model= self.raw_file_type,use_swap = True)
  File "C:\Users\proud\AppData\Roaming\Python\Python39\site-packages\echopype\utils\prov.py", line 215, in inner
    dataobj = func(*args, **kwargs)
  File "C:\Users\proud\AppData\Roaming\Python\Python39\site-packages\echopype\convert\api.py", line 442, in open_raw
    p2z.datagram_to_zarr(max_mb=max_mb)
  File "C:\Users\proud\AppData\Roaming\Python\Python39\site-packages\echopype\convert\parsed_to_zarr_ek60.py", line 289, in datagram_to_zarr
    self._write_power(df=self.datagram_df, max_mb=max_mb)
  File "C:\Users\proud\AppData\Roaming\Python\Python39\site-packages\echopype\convert\parsed_to_zarr_ek60.py", line 89, in _write_power
    self.write_df_column(
  File "C:\Users\proud\AppData\Roaming\Python\Python39\site-packages\echopype\convert\parsed_to_zarr.py", line 388, in write_df_column
    self.write_chunks(pd_series, zarr_grp, is_array, chunks, chunk_shape)
  File "C:\Users\proud\AppData\Roaming\Python\Python39\site-packages\echopype\convert\parsed_to_zarr.py", line 326, in write_chunks
    full_array = zarr_grp.array(
  ...

TypeError: Object of type int32 is not JSON serializable

Are you experiencing the same problem?

emiliom, 2023-06-05:
I don't think we've run into this. Can you report the version of echopype you are using and how you installed echopype (pip or conda)? Also, can you share a small file that produces this error?

leewujung, 2023-10-14:
Since the entire mechanism of `use_swap` is being refactored and we've not heard back on this, we'll close it for now. @rolandproud : if you run into the same issue later after the next release (v0.8.2) feel free to reopen this.

closed 2023-10-14 by leewujung