echostack-org/echopype #1062: Add descriptions and a diagram to docs documenting vertical offsets (water level, etc)

emiliom, 2023-06-11:
From @leewujung : For the sake of clarity, could you add the definition of `water_level` and `vertical_offset` with a simple diagram of which one means which for an underwater vehicle and a ship? This should actually be in the docs, but if we have something parked here it'll be easy to plug them in afterwards.

emiliom, 2023-08-17:
Refer to #1051 for related discussions.

leewujung, 2023-08-21:
I've restructured the TOC to have a Data processing section with 2 sections:
- `data-proc-func.ipynb`: providing usage examples, especially for `cal_params` and `env_params` for calibration
- `data-proc-additional.ipynb`: a one-stop-shopping place for additional details for processing functions 

One could argue that the `cal_params` and `env_params` details should be in the `-additional` page, but since I feel the `-func` page could also use some more expansion, I decided to hold off on any further changes.

I think the `-additional` page may be the place to go for the z-axis variables and combinations, and to include an illustration.

leewujung, 2023-08-25:
This is now addressed in #1133!

closed 2023-08-25 by leewujung