echostack-org/echopype #1074: Update installation documentation to use mamba

leewujung, 2023-07-12:
Many of us have been using mamba to install echopype for some time now, and since conda almost always takes forever or never finishes solving, it seems time for us to update the installation documentation to just use mamba.

Thoughts? @emiliom @lsetiawan

emiliom, 2023-07-12:
Agreed.

Here are the relevant sections in the docs:
- https://echopype.readthedocs.io/en/stable/contributing.html#installation-for-echopype-development
- https://echopype.readthedocs.io/en/stable/installation.html#installation

lsetiawan, 2023-07-17:
From scipy, I've learned that now you can actually just install the `libmamba` solver and configure that, so that when you're using conda, it will just use that solver, that way users don't need to learn much about mamba and get confused, especially if they're fairly new to the python ecosystem: https://conda.github.io/conda-libmamba-solver/getting-started/

Here's a nice article from Anaconda about it: https://www.anaconda.com/blog/conda-is-fast-now

leewujung, 2023-07-24:
Addressed in #1084.

closed 2023-07-24 by leewujung