echostack-org/echopype #1087: my files don't convert

Cristia7, 2023-07-24:
the same type files
a part can convert with more ping another part can't with less ping
code as follow

`from echopype import open_raw
import echopype.visualize as epviz
import matplotlib.pyplot as plt

ed = open_raw(raw_file="file_name.raw",sonar_model='EK80')
epviz.create_echogram(ed,vmax=-80, vmin=-200,channel='WBT 738177-15 ES38-7_ES')
plt.show()`

leewujung, 2023-07-24:
@Cristia7 : I saw that you created 3 issues in a roll but they do not provide enough information for us to help you.
Please go back to the (issue template)[https://github.com/OSOceanAcoustics/echopype/issues/new?assignees=&labels=my_files_don%27t_convert&projects=&template=my-files-don-t-convert.md&title=%5BFile+conversion%5D+SUSPECT_CAUSE} and fill the required info either in a new issue or put the required info here.

leewujung, 2023-07-25:
@Cristia7: No problem, thanks for the added info. Could you update to v0.7.1? Once you get the EchoData object (`ed`), try:
```
ed["Sonar/Beam_group1"]["backscatter_r"].sel(channel="WBT 738177-15 ES38-7_ES").plot(x="ping_time")
```
Do you get anything? If so we can look into if the viz module has some bugs (and we also need to update that part of documentation...)

Cristia7, 2023-07-26:
I have updated to 0.7.1 and  get `ed` after that run this code
 `ed["Sonar/Beam_group1"] ["backscatter_r"]. sel (channel="WBT 738177-15 ES38-7-ES"). plot (x="ping_time") `
 but The problem still exists
I think the problem arises in this part: `ed=open_raw(raw_file='files_name',sonar_model='EK80')`
because after run this code to obtain `ed` 
I also run 
```
ed.to_netcdf(save_path='files_name')
ed.to_zarr(save_path='files_name')
```
There is no correct information in the obtained NC file and zarr file. so I suspect that the `open_raw` part is not  obtain the `.raw`  information correctly

Cristia7, 2023-07-26:
I have tried other files that can be interpret
`.nc .zarr `files are right . both `ed["Sonar/Beam_group1"] ["backscatter_r"]. sel (channel="WBT 738177-15 ES38-7-ES"). plot (x="ping_time") `and `epviz.create_echogram(ed,channel='WBT 738177-15 ES38-7_ES')`able to draw correct images

leewujung, 2023-07-26:
> I have updated to 0.7.1 and get `ed` after that run this code
> `ed["Sonar/Beam_group1"] ["backscatter_r"]. sel (channel="WBT 738177-15 ES38-7-ES"). plot (x="ping_time") `
> but The problem still exists

Could you elaborate on what you meant by but the problem still exists? What is the problem here?

leewujung, 2023-07-26:
Well, that's what `x="ping_time"` is for and it's commented out in your code.

leewujung, 2023-07-26:
Also, since the values are so small, take a log would be useful:
```
np.log(ed["Sonar/Beam_group1"] ["backscatter_r"]. sel (channel="WBT 738177-15 ES38-7-ES")). plot (x="ping_time")
```

Cristia7, 2023-07-26:
When `x="ping_time"` is included in my code, this situation occurs
[image]

TypeError: Axes.hist() got multiple values for argument 'x'

Cristia7, 2023-07-26:
When x is included in my code, this situation occurs

[image]

TypeError: Axes.hist() got multiple values for argument 'x'

leewujung, 2024-02-25:
@Cristia7 : I am not sure what the exact problem is for this issue, as it does not seem to be a code problem but in plotting. Any update?

leewujung, 2024-05-01:
Closing this now due to inaction. Feel free to open up new issues if you still run into problems.

closed 2024-05-01 by leewujung