echostack-org/echopype #1092: zarr backend encoding parameters

jmjech, 2023-07-26:
Hi,
I'm attempting to combine three .nc files using the command
ed = ep.combine_echodata(edlist, overwrite=True) and I get the following error messages:

>>> ed=ep.combine_echodata(edlist, overwrite=True)
/home/mjech/.local/lib/python3.10/site-packages/distributed/node.py:182: UserWarning: Port 8787 is already in use.
Perhaps you already have a cluster running?
Hosting the HTTP server on port 37729 instead
  warnings.warn(
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mjech/.local/lib/python3.10/site-packages/echopype/echodata/combine.py", line 597, in combine_echodata
    ed_comb = comb.combine(
  File "/home/mjech/.local/lib/python3.10/site-packages/echopype/echodata/zarr_combine.py", line 973, in combine
    const_names = self._append_ds_list_to_zarr(
  File "/home/mjech/.local/lib/python3.10/site-packages/echopype/echodata/zarr_combine.py", line 630, in _append_ds_list_to_zarr
    ds_lazy.to_zarr(
  File "/home/mjech/.local/lib/python3.10/site-packages/xarray/core/dataset.py", line 2236, in to_zarr
    return to_zarr(  # type: ignore[call-overload,misc]
  File "/home/mjech/.local/lib/python3.10/site-packages/xarray/backends/api.py", line 1687, in to_zarr
    dump_to_store(dataset, zstore, writer, encoding=encoding)
  File "/home/mjech/.local/lib/python3.10/site-packages/xarray/backends/api.py", line 1293, in dump_to_store
    store.store(variables, attrs, check_encoding, writer, unlimited_dims=unlimited_dims)
  File "/home/mjech/.local/lib/python3.10/site-packages/xarray/backends/zarr.py", line 624, in store
    self.set_variables(
  File "/home/mjech/.local/lib/python3.10/site-packages/xarray/backends/zarr.py", line 671, in set_variables
    encoding = extract_zarr_variable_encoding(
  File "/home/mjech/.local/lib/python3.10/site-packages/xarray/backends/zarr.py", line 271, in extract_zarr_variable_encoding
    raise ValueError(
ValueError: unexpected encoding parameters for zarr backend:  ['zlib', 'shuffle', 'complevel', 'fletcher32', 'contiguous', 'chunksizes']
>>> 

The full code is:
import echopype as ep
from pathlib import Path
#from dask.distributed import Client
import xarray as xr
import numpy as np
import pandas as pd

# the Client gives lots of error messages, so I don't use it for now
#client = Client()

# get multiple files
dataDirectory = Path('/home/mjech/Desktop/butterfish/netCDF4_Files')
filelist = []
# sort the filelist
for f in sorted(Path(dataDirectory).glob('**/*.nc')):
    print('.nc file: ', f)
    filelist.append(f)

edlist = []
for f in filelist:
    edlist.append(ep.open_converted(str(f)))
#ed = ep.combine_echodata(edlist, client=client)
ed = ep.combine_echodata(edlist, overwrite=True)

I'm using python 3.10.6
Ubuntu 22.04.2 LTS
I just installed echopype on 25 July 2023, so I think it's version 0.7.1

I'm sure I'm forgetting something, so please let me know what other information you need.

Thanks!

leewujung, 2023-07-26:
@jmjech : Thanks for reporting this. This should have been resolved in the `dev` branch where we have overhauled `combine_echodata`. We are trying our best to do a new release by end of this week, but for the time being you could try `pip install git+https://github.com/OSOceanAcoustics/echopype.git@dev` to get the latest that should be free of this problem (but let us know if you still find bugs!)

jmjech, 2023-07-26:
Thanks! Installing the dev version worked.
on to the next bug :)

leewujung, 2023-07-27:
> Installing the dev version worked.

Great!

> on to the next bug :)

Yes, please keep'em coming! :)

closed 2023-08-12 by leewujung