echostack-org/echopype #1095: Support for ER60 Simrad .raw files ?

Tchiimy, 2023-07-28:
Hello, 

In my project I need to read and use data from an echo-sounder. The data seems readable with Simrad ER60. I would like to be able to read the .raw files, for this I have checked this [documentation](https://www.manualslib.com/manual/381563/Simrad-Er60-Rev-B.html?page=142#manual) that presents the data formats. 

I don't have that much experience in .raw files reading but I suppose that If I have the data formats I could use some python library to create a function to read it ? 

Do you have any tips on how to do it or already have tried it ?

If you want I can give you a random .raw file I made to give you a better idea of the file format.
[TRIFREC_B859_120KHZ_200W_300MS_80M_-D20220915-T100451 (2).zip](https://github.com/OSOceanAcoustics/echopype/files/12196060/TRIFREC_B859_120KHZ_200W_300MS_80M_-D20220915-T100451.2.zip)

Alban

Tchiimy, 2023-10-18:
Sorry for the late answer ! 

You were right, echopype worked fine with these files ! I had trouble installing the library so I didn't thought it would be compatible with my system, but after your message I regave it a try and it worked thanks !!

closed 2023-10-18 by Tchiimy

leewujung, 2023-10-18:
Oh awesome! If you have some inputs on how to improve the installation instructions, let us know as well!