echostack-org/echopype #1140: CI is not correctly raising the exit code from test

lsetiawan, 2023-08-29:
## Overview

I noticed that currently in `dev` branch the tests are "passing". However, looking at the actual output log [here](https://github.com/OSOceanAcoustics/echopype/actions/runs/5992404435/job/16251756753#step:11:4310) There are actual some failures in the test, but it's still showing green checkmark.

I think the issue here is that using `tee` to capture the logs is not allowing the actual test exit code to be raised, but rather the success of writing to log file. Need to find a better way to do this so that we can both capture the log and raise the appropriate error code.

emiliom, 2023-08-30:
I noticed this behavior in a recent PR! But it's clearly not consistent, since we do get CI failures due to test failures

leewujung, 2023-08-30:
Thanks for adding this issue. I noticed that too and ran all tests locally before merging.