echostack-org/echopype #1163: Explore inefficiencies in `cal_params.py`

lsetiawan, 2023-09-13:
There's a potential inefficiency in EK60 and EK80 'cw' data. See code below of where this might be.

https://github.com/OSOceanAcoustics/echopype/blob/3ca028f04a4fd3643bb0ad350dc09a699883ee1e/echopype/calibrate/cal_params.py#L291-L297

anantmittal, 2023-10-03:
Command to run the tests and capture the stdout:
`python .ci_helpers/run-test.py --local --pytest-args="-vvrP" echopype/calibrate/cal_params.py`

lsetiawan, 2023-10-03:
## Pytest Run

```
pytest -vvrP echopype/tests/calibrate/test_cal_params.py::test_get_vend_cal_params_power
```

leewujung, 2023-10-31:
This issue now is captured more fully in #1200. @lsetiawan @anantmittal : you may wanna close this and copy-paste the above comments there.

closed 2023-11-01 by lsetiawan