echostack-org/echopype #1165: Check to see if this masking `xr.where` parallelize

lsetiawan, 2023-09-13:
https://github.com/OSOceanAcoustics/echopype/blob/3ca028f04a4fd3643bb0ad350dc09a699883ee1e/echopype/mask/api.py#L500

anantmittal, 2023-10-03:
Command to run the tests and capture the stdout:
`python .ci_helpers/run-test.py --local --pytest-args="-vvrP" echopype/mask/api.py`

leewujung, 2023-10-24:
@lsetiawan : the code you pointed to above was actually not the one we were discussing 😬

Below are the 2 potential places where the calibration code (`compute_Sv`) code can have bottleneck:

- in `get_vend_cal_params_power` where a broadcasting is probably used to get a data variable into the "right" shape based on indexing:
https://github.com/OSOceanAcoustics/echopype/blob/593a7e173dffd34858a7587cf2f31633a23a841c/echopype/calibrate/cal_params.py#L299-L304

- in `harmonize_env_param_time` where there is either a check for all the timestamps or an interpolation:
https://github.com/OSOceanAcoustics/echopype/blob/593a7e173dffd34858a7587cf2f31633a23a841c/echopype/calibrate/env_params.py#L53-L66

leewujung, 2024-02-25:
I'll close this now, since my comment above is replicated in #1200.

PR #1198 (2024-02-25): Optimize Frequency Differencing with Dask
Fixes #1165
closed 2024-02-25 by leewujung