echostack-org/echopype #1171: Add storage for XML parameters from >1 phases in AZFP

leewujung, 2023-09-18:
With the new XML parser in #1135, we can now handle more than 1 phase in AZFP data. However, currently only the phase 1 parameters parsed from XML are stored in the `Vendor_specific` group. I think we can add an additional dimension `phase` to the XML parameters that are under the different phases to store all the information.

emiliom, 2023-09-26:
Ideally, we should get an AZFP file that does have >1 phase, for testing. Currently it doesn't look like we do.

leewujung, 2023-09-26:
I added one actually, it's already in the CI. I've pointed @praneethratna to it. It is the one Steve Pearce provided.

emiliom, 2023-09-26:
I just checked the `CI_test_data` Google Drive folder. Are you referring to `azfp/23081211.XML`? If so, Steve only sent us the XML file, not the data (01A) file. The folder contains three XML files but only two .01A files.

leewujung, 2023-09-26:
Yes, that is the file. I think for testing parsing the XML that is sufficient, but maybe you're thinking about going all the way to the `set_vendor` part? Maybe we can ask Steve for an example 01A file?

emiliom, 2023-09-26:
> but maybe you're thinking about going all the way to the set_vendor part? 

Yup

> Maybe we can ask Steve for an example 01A file?

Yeah, hopefully he has the .01A file that goes with the XML file, or another complete set.

praneethratna, 2023-09-26:
> I added one actually, it's already in the CI. I've pointed @praneethratna to it. It is the one Steve Pearce provided.

Yes, I'll actually change the test to use this new file for better tests of new code.

closed 2023-10-07 by praneethratna