echostack-org/echopype #1176: compute_Sv return 'NAN'

wangtaojiezi, 2023-09-19:
**General description of problem**
I am using a 'compute_Sv',there is' nan 'in the 'Sv' and echo_range, and the position where nan exists does not correspond. Is this correct? What is the meaning of this' nan' and how to handle 'nan'.

**Computing environment**
Echopype version: 0.8.1
Operating system:Windows 11

**### Minimum example**
The following code reproduces the issue I encountered:
```
range_kwargs1 = {"waveform_mode": "BB", "encode_mode": "complex"}
ds_Sv = ep.calibrate.compute_Sv(ed, **range_kwargs1).compute()
print(ds_Sv)
```

**outputdata**
The data samples are as follows：
sv：[NaN, NaN, -190.77, -47.6, -41.16,NaN, NaN, NaN, NaN,]
echo_range: [0.01, 0.02, 0.03, 0.04, 0.05, 0.06,NaN, NaN, [NaN]

[Uploading outputdata.txt…]()

I am currently confused and hope to receive your help

leewujung, 2023-10-25:
@wangtaojiezi: Does your data only have 1 channel? Could you do a printout of `ds_Sv["Sv"]` and `echo_range` in a gist or text file or something similar that shows the dimension and length of these variables with some data snippet? It's hard to determine what you showed without more information. Thanks.

leewujung, 2024-03-28:
This may be of the same source as #1260? Not sure, but since there's no further info here and it's been a few months, I'll close this now. @wangtaojiezi we'll ping you once #1260 is addressed. If you continue to run into problem afterwards feel free to raise a new issue.

closed 2024-03-28 by leewujung