echostack-org/echopype #1179: Replace the current `parsed2zarr` mechanism

leewujung, 2023-09-21:
This issue is to pick up from #1070 to replace the current `parsed2zarr` mechanism to ensure code is bug-free and easy to maintain in the future.

I am referencing my https://github.com/OSOceanAcoustics/echopype/pull/1070#issuecomment-1730079171 here:

> The reason why i think the current complicated approach is unnecessary is because there are 2 stages of padding that is happening:
> 
> the "padding" for each channel to become rectangular (so to pad the shorter pings so that all pings within a channel have the same length) is in pad_shorter_ping. That is the first NaN-padding expansion.
> The second NaN-padding expansion is when some channels lack a certain type of data, so NaN is padded for the entire "slice" of that channel to match the size of the other channels.
> Expansion 1 happens for all data without xarray/zarr/dask (which can potentially be sped up or make more efficient?). After that, if we take note of the shape of variables in self.ping_data_dict["angle"][ch_id] for each ch_id , we would know exactly whether or not expansion 2 is needed. There, we can decide what is the strategy to use to write to the zarr store -- can we append sequentially along segments of ping_time since we know all the pings are same length? and for the channels that lack any data at all, creating a NaN slice for each sections (along ping_time segments) would likely let us hover under the memory limit.
> 

@lsetiawan and I have done some investigation and determined a clean path forward.

leewujung, 2023-09-27:
Notes while going through code:

Right now the power, angle, and complex samples are all directed to be saved in `parser.zarr_datagrams` after `parser.parse_raw` -- I think this is unnecessary and messy. It helps with data integrity and therefore later maintenance to keep these variables along with others in `parser.ping_data_dict`.
- To change this, remove the handling specific to `zarr_vars` in all occurrence of the use of `_append_channel_ping_data`. 
- Better to keep a declaration up front for which variables should be handled with `swap=True` and select from `parser.ping_data_dict`

lsetiawan, 2023-11-06:
This Issue is related to #966

closed 2023-11-07 by lsetiawan