echostack-org/echopype #1181: Parse AZFP pressure data when available

emiliom, 2023-09-24:
AZFP deployments can contain a pressure sensor. However, the echopype AZFP parser currently does not look for and read pressure data, as it does for temperature. Add this.

When read, it should be used to populate `Platform.vertical_offset` while also setting `Platform.water_level` to 0. See https://echopype.readthedocs.io/en/stable/data-proc-additional.html#vertical-coordinate-z-axis-variables

emiliom, 2023-10-15:
After parsing the pressure data, the data array needs to be written into the EchoData object.

I think there are two steps here:
1. Store `pressure` in the `Environment` group, via the `set_env` class method. Populate its variable attributes. Base this on what's done for temperature, but modifying attribute values to correspond to pressure. I can provide those attributes here, later.
2. Use `pressure` and a standard seawater equation ("equations of state") to calculate depth (meters). Then set `Platform.vertical_offset` to this depth array, multiplied by -1 (see [here](https://echopype.readthedocs.io/en/stable/data-proc-additional.html#vertical-coordinate-z-axis-variables)). `LoadAZFP.m` has this conversion in the `computeDepth` function. It doesn't look like we have a seawater density function in echopype, but we can incorporate one. The "ideal" package would be https://github.com/TEOS-10/GSW-Python, but unfortunately it depends on a C library. Note that `GSW-Python` already includes a function to go from pressure to depth (see [gsw.conversions.z_from_p](https://teos-10.github.io/GSW-Python/conversions.html)), but it requires at least one parameter that may not be available: latitude. @leewujung should we use simplifications for density (we'll have to use a default salinity anyways) and omit a latitude adjustment?

emiliom, 2023-10-23:
PR #1189 addressed item 1 above: parsing and storing pressure data, plus tests.

For item 2, we'll need to decide what conversion formula / algorithm / library to use to convert from pressure to depth.

emiliom, 2023-10-24:
For reference, here's the functionality used by the AZFP Matlab toolbox to calculate density (The links below are to a private repo):

- In `LoadAZFP.m`, [computeDepth function](https://github.com/OSOceanAcoustics/AZFP_matlab/blob/main/matlab_toolbox/LoadAZFP.m#L540):
  ```matlab
  g = 9.815;
  dens = sw_dens(Parameters.Salinity,T,Pressure);
  Depth = Pressure*10000 / (dens * g);
  ```
- `sw_dens` is found in the same toolbox, in [sw_dens.m](https://github.com/OSOceanAcoustics/AZFP_matlab/blob/main/matlab_toolbox/sw_dens.m). It's based on the "Density of Sea Water using UNESCO 1983 (EOS 80) polynomial." It calls two other functions, [sw_dens0.m](https://github.com/OSOceanAcoustics/AZFP_matlab/blob/main/matlab_toolbox/sw_dens0.m) and [sw_seck.m](https://github.com/OSOceanAcoustics/AZFP_matlab/blob/main/matlab_toolbox/sw_seck.m)

emiliom, 2023-10-24:
Hmm, the `python-gsw` function [z_from_p](https://github.com/TEOS-10/python-gsw/blob/7d6ebe8114c5d8b4a64268d36100a70e226afaf6/gsw/gibbs/conversions.py#L1483) doesn't require a density calculation, or salinity. It does require latitude, naturally, but since AZFP data will lack latitude, we could input a mid-range value, say, 45 degrees.

emiliom, 2023-10-26:
@praneethratna Ok, I now have a recommendation for implementing the conversion from pressure to depth. Let's use the [UNESCO 1983](https://unesdoc.unesco.org/ark:/48223/pf0000059832) formulation (pages 25-27). [This site](http://the-mostly.ru/misc/depth_from_pressure_online_calculator.html) lists the equations nicely and includes a calculator. We'll make the standard seawater assumption used in that site: salinity of 35 PSU and temperature of 0 C. That makes the $&Delta;D$ term in UNESCO 1983 equation 25 become zero. That equation is then identical as eq. 3 in the other site.

Let's implement this as a new function `depth_from_pressure` in `utils/misc.py`. The arguments will be:
- pressure (numpy array)
- latitude (scalar), with a default of 30
- atm_pres_surf (scalar), with a default of 0 (atmospheric pressure at the surface)

In the function, `atm_pres_surf` will be subtracted from the pressure array.

The function returns a depth numpy array.

ctuguinay, 2024-04-24:
This should be closed by the merging of #1207?

closed 2024-04-24 by ctuguinay