echostack-org/echopype #1200: Improve potential bottlenecks in `compute_Sv`

lsetiawan, 2023-10-26:
Below are the 2 potential places where the calibration code (`compute_Sv`) code can have bottleneck:

- in `get_vend_cal_params_power` where a broadcasting is probably used to get a data variable into the "right" shape based on indexing:
https://github.com/OSOceanAcoustics/echopype/blob/593a7e173dffd34858a7587cf2f31633a23a841c/echopype/calibrate/cal_params.py#L299-L304

- in `harmonize_env_param_time` where there is either a check for all the timestamps or an interpolation:
https://github.com/OSOceanAcoustics/echopype/blob/593a7e173dffd34858a7587cf2f31633a23a841c/echopype/calibrate/env_params.py#L53-L66

_Originally posted by @leewujung in https://github.com/OSOceanAcoustics/echopype/issues/1165#issuecomment-1776360813_

anantmittal, 2023-11-03:
```
pytest -vvrP echopype/tests/calibrate/test_cal_params.py::test_get_vend_cal_params_power
pytest -vvrP echopype/tests/calibrate/test_env_params.py::test_harmonize_env_param_time
```

anantmittal, 2023-11-25:
@leewujung Could you explain a bit more about the potential bottleneck in `get_vend_cal_params_power `? Do we want to parallelize specific operations using `dask`? Or do we want to refactor the code using `numpy/xarray` itself to improve runtime?

anantmittal, 2023-11-25:
@leewujung Similar question for improvement in `harmonize_env_param_time `. Are we looking into parallelizing specific (e.g., np.all) operations?

leewujung, 2023-11-26:
> Do we want to parallelize specific operations using `dask`? Or do we want to refactor the code using `numpy/xarray` itself to improve runtime?

I think it is both. We want to enable handling data with lazy-loaded arrays if it's not already, and ensure that the operations are distributed efficiently when dask supports them (and if not, see where dask is heading on the ops, since these ops are not very specific).

In `get_vend_cal_params_power`, I worry that `expand_dims` and to a lesser degree `sortby` is expensive.

In `harmonize_env_param_time`, the `np.all` in one of the conditions may be slow if time dimension is large, and `.interp` may also be slow.

anantmittal, 2023-11-30:
For `harmonize_env_param_time `,

On running some experiments, `np.all` appears to be the fastest way when using just `numpy`. This [link](https://stackoverflow.com/questions/10580676/comparing-two-numpy-arrays-for-equality-element-wise) also shares the same thought.

leewujung, 2024-02-25:
This `harmonize_env_param_time` component was addressed in #1235.

leewujung, 2024-05-01:
We can close this now with #1235 and #1285 merged.