echostack-org/echopype #1210: TypeError: File contains changing transmit_duration_nominal!

bjarnekvae, 2023-11-03:
Hi,

I am getting the following error while trying to open some EK80 data:
TypeError: File contains changing transmit_duration_nominal!

The error originates from:
https://github.com/OSOceanAcoustics/echopype/blob/0de62b0b36c0c08417723c9702f86689f1bd1e5b/echopype/calibrate/ek80_complex.py#L245

Here are some details about the RAW-file format version seems to be v1.27.

What can I do to fix this error?

Also, previously I had issues with the following line:
https://github.com/OSOceanAcoustics/echopype/blob/0de62b0b36c0c08417723c9702f86689f1bd1e5b/echopype/convert/utils/ek_raw_parsers.py#L1638

Giving me this error:
ValueError: When changing to a larger dtype, its size must be a divisor of the total size in bytes of the last axis of the array.

However, I fixed this by changeing this line to:
`data["complex"] = data["complex"].astype(np.complex64)`

praneethratna, 2023-12-18:
Hey @bjarnekvae Thanks for reporting this! could you share the test file so that we can look into this more closely?

bjarnekvae, 2023-12-18:
Hi, and tanks for the reply! 

Sorry for not updating this, but it turned out that the issue was a unusual combination of hardware and software for echo sounder data acquisition. Nothing to do with EchoPype

closed 2023-12-18 by bjarnekvae