echostack-org/echopype #1211: Improve `estimate_noise` binning routine

lsetiawan, 2023-11-06:
## Overview

This issue follows up on Issue https://github.com/OSOceanAcoustics/echopype/issues/1028. Improvements in 2D binning routines have been implemented for both `compute_MVBS`

MohamedNasser8, 2024-04-12:
Hello @lsetiawan,
As I understand it, `ping_time` can refer to actual timestamps or simply be a sequence of indices. We need to account for both scenarios: when `ping_time` represents real-time data and when it represents indices. I am making sure to handle both cases appropriately to ensure accurate results.

ctuguinay, 2024-07-16:
Closing this as coarsen itself is very fast and probably does not need the Flox algorithms to speed up this process.

closed 2024-07-16 by ctuguinay