echostack-org/echopype #1212: Failure to perform `compute_Sv` with dask inputs

lsetiawan, 2023-11-07:
## Overview

Currently, `compute_Sv` does not work with a merged Sv file (`raw_converted_combined/2013/x0097_2_wt_20130824_232034_f0031.zarr`) when opened as dask arrays. Worker memory get's overwhelmed and even spilling to disk by dask won't cut it. This is the non descriptive dask traceback:

```console
2023-11-06 17:00:16,536 - distributed.worker.memory - WARNING - Unmanaged memory use is high. This may indicate a memory leak or the memory may not be released to the OS; see https://distributed.dask.org/en/latest/worker-memory.html#memory-not-released-back-to-the-os for more information. -- Unmanaged memory: 11.37 GiB -- Worker memory limit: 16.00 GiB
2023-11-06 17:00:16,878 - distributed.worker.memory - WARNING - Worker is at 80% memory usage. Pausing worker.  Process memory: 12.85 GiB -- Worker memory limit: 16.00 GiB
2023-11-06 17:00:16,922 - distributed.worker.memory - WARNING - Worker is at 79% memory usage. Resuming worker. Process memory: 12.70 GiB -- Worker memory limit: 16.00 GiB
...
RuntimeError: Cannot remove replica of ('transpose-4cd487e9eb8037d494b448b470058650', 0, 0, 0) while ('rechunk-split-rechunk-merge-78880722c4e511b3105b467cdecc2000', 0, 22, 2) in state 'executing'.
RuntimeError: Cannot remove replica of ('transpose-4cd487e9eb8037d494b448b470058650', 0, 0, 0) while ('rechunk-split-rechunk-merge-78880722c4e511b3105b467cdecc2000', 0, 22, 2) in state 'executing'.
```

The above traceback is not very useful, so I tried to perform `compute_Sv` without using dask and that seemed to work and I was able to profile the memory usage. It seems like there are a few spots that huge spikes in memory occurs.

<details>
<summary>Memory Line by Line Profile</summary>
<br>

```
Filename: /Users/lsetiawan/Repos/SSEC/echopype/echopype/calibrate/calibrate_ek.py

Line #    Mem usage    Increment  Occurrences   Line Contents
=============================================================
   137    222.6 MiB    222.6 MiB           1       def __init__(self, echodata: EchoData, env_params, cal_params, ecs_file, **kwargs):
   138    222.6 MiB      0.0 MiB           1           super().__init__(echodata, env_params, cal_params, ecs_file)
   139                                         
   140                                                 # Set sonar_type and waveform/encode mode
   141    222.6 MiB      0.0 MiB           1           self.sonar_type = "EK60"
   142                                         
   143                                                 # Set cal type
   144    222.6 MiB      0.0 MiB           1           self.waveform_mode = "CW"
   145    222.6 MiB      0.0 MiB           1           self.encode_mode = "power"
   146                                         
   147                                                 # Get the right ed_beam_group for CW power samples
   148    222.6 MiB      0.0 MiB           2           self.ed_beam_group = retrieve_correct_beam_group(
   149    222.6 MiB      0.0 MiB           1               echodata=self.echodata, waveform_mode=self.waveform_mode, encode_mode=self.encode_mode
   150                                                 )
   151                                         
   152                                                 # Set the channels to calibrate
   153                                                 # For EK60 this is all channels
   154    222.6 MiB      0.0 MiB           1           self.chan_sel = self.echodata[self.ed_beam_group]["channel"]
   155    222.6 MiB      0.0 MiB           1           beam = self.echodata[self.ed_beam_group]
   156                                         
   157                                                 # Convert env_params and cal_params if self.ecs_file exists
   158                                                 # Note a warning if thrown out in CalibrateBase.__init__
   159                                                 # to let user know cal_params and env_params are ignored if ecs_file is provided
   160    222.6 MiB      0.0 MiB           1           if self.ecs_file is not None:  # also means self.ecs_dict != {}
   161                                                     ds_env_tmp, ds_cal_tmp, _ = ecs_ev2ep(self.ecs_dict, "EK60")
   162                                                     self.cal_params = ecs_ds2dict(
   163                                                         conform_channel_order(ds_cal_tmp, beam["frequency_nominal"])
   164                                                     )
   165                                                     self.env_params = ecs_ds2dict(
   166                                                         conform_channel_order(ds_env_tmp, beam["frequency_nominal"])
   167                                                     )
   168                                         
   169                                                 # Regardless of the source cal and env params,
   170                                                 # go through the same sanitization and organization process
   171    229.0 MiB      6.4 MiB           2           self.env_params = get_env_params_EK(
   172    222.6 MiB      0.0 MiB           1               sonar_type=self.sonar_type,
   173    222.6 MiB      0.0 MiB           1               beam=self.echodata[self.ed_beam_group],
   174    222.6 MiB      0.0 MiB           1               env=self.echodata["Environment"],
   175    222.6 MiB      0.0 MiB           1               user_dict=self.env_params,
   176                                                 )
   177    280.5 MiB     51.5 MiB           2           self.cal_params = get_cal_params_EK(
   178    229.0 MiB      0.0 MiB           1               waveform_mode=self.waveform_mode,
   179    229.0 MiB      0.0 MiB           1               freq_center=beam["frequency_nominal"],
   180    229.0 MiB      0.0 MiB           1               beam=beam,
   181    229.0 MiB      0.0 MiB           1               vend=self.echodata["Vendor_specific"],
   182    229.0 MiB      0.0 MiB           1               user_dict=self.cal_params,
[... 280 lines cut ...]
```
190   1511.1 MiB  -4904.3 MiB           1       range_meter = range_meter.where(valid_idx)
```

This is then cleaned up by a where operation on the `range_meter` variable calculated above.

**echopype/calibrate/calibrate_ek.py::_cal_power_samples**

```
74   7130.0 MiB   7130.0 MiB           2           tvg_mod_range = range_mod_TVG_EK(
75    946.3 MiB      0.0 MiB           1               self.echodata, self.ed_beam_group, self.range_meter, sound_speed
76                                                 )
...
80   6107.8 MiB   5454.9 MiB           1           absorption_loss = 2 * absorption * tvg_mod_range
```

Once `compute_range` is finished, the process continues to calling `_cal_power_samples` to perform the `Sv` computation. As seen in the snippet above, `range_mod_TVG_EK` increased the memory usage `7130.0 MiB`... that's around `7GB`! Then another spike of `5454.9 MiB` happens again during the computation above.

**echopype/calibrate/range.py::range_mod_TVG_EK**

```
229   7129.7 MiB   6174.4 MiB           1           range_meter = range_meter - mod_Ex60()
```

Digging into `range_mod_TVG_EK` function, we can see that the substraction b/w `range_meter` and `mod_Ex60` function result blows up the memory here.

lsetiawan, 2023-11-07:
## Investigation

After some initial investigate, I found that the first computation of `beam["range_sample"] * beam["sample_interval"] * sound_speed / 2` is really problematic. We can re-create a very simple example with just a few scaffolding:

```python
from distributed import Client
import dask.array
import xarray as xr
import numpy as np

# Use 2 workers so each gets more memory
client = Client(n_workers=2)

# Same shape as the actual data
ch, pt, rs = 5, 46187, 3957
sample_interval = xr.DataArray(dask.array.ones((ch, pt)), name="sample_interval", coords={"ch": np.arange(ch), "pt": np.arange(pt)})
sound_speed = xr.DataArray(dask.array.ones((ch, pt)) * 1053, name="sound_speed", coords={"ch": np.arange(ch), "pt": np.arange(pt)})
range_sample = xr.DataArray(dask.array.from_array(np.arange(rs)), name="range_sample", coords={"rs": np.arange(rs)})
res = range_sample * sample_interval * sound_speed / 2
res.compute()
```

The above code WILL FAIL with similar error and memory spilling and blowup seen in the dask dashboard:

```console
2023-11-06 17:17:59,276 - distributed.worker.memory - WARNING - Unmanaged memory use is high. This may indicate a memory leak or the memory may not be released to the OS; see https://distributed.dask.org/en/latest/worker-memory.html#memory-not-released-back-to-the-os for more information. -- Unmanaged memory: 11.84 GiB -- Worker memory limit: 16.00 GiB
2023-11-06 17:17:59,495 - distributed.worker.memory - WARNING - Worker is at 83% memory usage. Pausing worker.  Process memory: 13.36 GiB -- Worker memory limit: 16.00 GiB
2023-11-06 17:17:59,663 - distributed.worker.memory - WARNING - Worker is at 43% memory usage. Resuming worker. Process memory: 6.91 GiB -- Worker memory limit: 16.00 GiB
```

If we perform the same snippet of code with the following shape:

```python
ch, pt, rs = 5, 20000, 2000
```

the computation will finish just fine.

ctuguinay, 2024-05-24:
So the most obvious problem here comes from `range_sample` multiplied by either `sound_speed` or `sample_interval`. This creates an array that has memory approximately 3975 times bigger than that of both `sample_interval` and `sound_speed`. Even though this is parallelized, I suspect that there's some excessive broadcasting going on that may overload worker memory. I'm still looking into this.

ctuguinay, 2024-06-05:
Using two workers doesn't seem to be helpful, since most of the memory will stick to just one worker here

leewujung, 2024-06-05:
I would suggest chunking along `ping_time`. `range_sample` is typically shorter for large dataset, since the depth setting usually does not get fiddled much, but `ping_time` can continue to grow.

Maybe look at arithmetic things that could simplify the ops would be helpful? like is it possible to avoid the expansion, etc.

ctuguinay, 2024-06-05:
Yeah, I tried just chunking on `ping_time`, and that seemed to scale well, speed-wise, but there's a large subset of task outputs that are stored in RAM and this builds up over time. I can also look for stuff that avoids expansion altogether.

leewujung, 2024-06-05:
Yeah, separating out the steps and looking at the graphs built for each step would likely be useful for figuring out exactly where the bottleneck is -- though I think you're already doing it! :)

ctuguinay, 2024-06-07:
I'll put the (potential) fix for this issue into a separate issue since the idea is extensive enough that it needs its own separate thing. To summarize, this issue is caused by the accumulation of task products that build up due to Dask waiting for every chunk to compute before setting the end array of `res.compute()`. To fix this accumulation problem, we can aim the computation at a Zarr store.

ctuguinay, 2024-07-16:
Closed by merging of #1331

closed 2024-07-16 by ctuguinay