echostack-org/echopype #1213: Move `complex` data type reshaping to parser rather than during rectangularize

lsetiawan, 2023-11-07:
## Overview

In PR https://github.com/OSOceanAcoustics/echopype/pull/1185, after the swap mechanism was refactored, @leewujung made a suggestion to port over the `complex` data type reshaping to the parser object rather than during a rectangularization step.

See TODO below:

https://github.com/OSOceanAcoustics/echopype/blob/499866f34266f1748c192e8d88a0931f39e8efa3/echopype/convert/parse_base.py#L294-L296

The referenced function is found below

https://github.com/OSOceanAcoustics/echopype/blob/499866f34266f1748c192e8d88a0931f39e8efa3/echopype/convert/utils/ek_raw_parsers.py#L1534-L1633

_Originally posted by @leewujung in https://github.com/OSOceanAcoustics/echopype/pull/1185#discussion_r1385160862_

leewujung, 2023-11-08:
Hey @praneethratna : Would you be willing to take this up, since you are familiar with the parser? I think [all test ci] would need to be used for the PR, to make sure all tests get run through, since the parser downstream is everything. Thanks!

praneethratna, 2023-11-08:
@leewujung sure yes! i'll take this up. Thanks!

closed 2023-11-16 by praneethratna