echostack-org/echopype #1220: Error upon trying calibration of Simrad EK80 .nc converted file: Key Error "transmit_type"

icaro-00, 2023-11-10:
Hello all!  I am currently trying to get sv values from Simrad EK80 echosounder. I converted the .raw file to .nc (netCDF4) files. I am now moving on the second step, where I am processing the .nc converted file and calibrate to get sv values. However, if I run the script provided, I run into the error:

KeyError: 'transmit_type'

Does anyone know how to proceed to fix it? Or am I missing some mid-steps prior to calibration? Or?

The code I run:

nc_path = r"C:\Users\Runner\Desktop\Echopype_v2.0\unpacked_files\20230925--D20230925-T083404.nc"  
echodata = ep.open_converted(nc_path)

ds_Sv = ep.calibrate.compute_Sv(echodata, waveform_mode="CW", encode_mode="complex")

I get this error:

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
~\anaconda3\envs\Echopype\lib\site-packages\xarray\core\dataset.py in ?(self, name)
   1446             variable = self._variables[name]
   1447         except KeyError:
-> 1448             _, name, variable = _get_virtual_variable(self._variables, name, self.dims)
   1449 

KeyError: 'transmit_type'

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_24036\2366883360.py in ?()
      1 nc_path = r"C:\Users\Runner\Desktop\Echopype_v2.0\unpacked_files\20230925--D20230925-T083404.nc"  # path to a converted nc file
      2 echodata = ep.open_converted(nc_path)  # create an EchoData object
      3 
      4 # For EK80 data, you need to specify waveform_mode and encode_mode
----> 5 ds_Sv = ep.calibrate.compute_Sv(echodata, waveform_mode="CW", encode_mode="complex")

~\anaconda3\envs\Echopype\lib\site-packages\echopype\calibrate\api.py in ?(echodata, **kwargs)
    202     The returned xr.Dataset will contain the variable `water_level` from the
    203     EchoData object provided, if it exists. If `water_level` is not returned,
    204     it must be set using `EchoData.update_platform()`.
    205     """
--> 206     return _compute_cal(cal_type="Sv", echodata=echodata, **kwargs)

~\anaconda3\envs\Echopype\lib\site-packages\echopype\calibrate\api.py in ?(cal_type, echodata, env_params, cal_params, ecs_file, waveform_mode, encode_mode)
     45                 "(encode_mode='power'). Calibration will be done on the power samples.",
     46             )
     47 
     48     # Set up calibration object
---> 49     cal_obj = CALIBRATOR[echodata.sonar_model](
     50         echodata,
     51         env_params=env_params,
     52         cal_params=cal_params,

~\anaconda3\envs\Echopype\lib\site-packages\echopype\calibrate\calibrate_ek.py in ?(self, echodata, env_params, cal_params, waveform_mode, encode_mode, ecs_file, **kwargs)
    231         self.encode_mode = encode_mode
    232         self.echodata = echodata
    233 
    234         # Get the right ed_beam_group given waveform and encode mode
--> 235         self.ed_beam_group = retrieve_correct_beam_group(
    236             echodata=self.echodata, waveform_mode=self.waveform_mode, encode_mode=self.encode_mode
    237         )
    238 

~\anaconda3\envs\Echopype\lib\site-packages\echopype\echodata\simrad.py in ?(echodata, waveform_mode, encode_mode)
    214         power_ed_group = _retrieve_correct_beam_group_EK60(echodata, waveform_mode, encode_mode)
    215 
    216     elif echodata.sonar_model in ["EK80", "ES80", "EA640"]:
    217         # check modes against data for EK80 and get power/complex EchoData groups
--> 218         power_ed_group, complex_ed_group = _retrieve_correct_beam_group_EK80(
    219             echodata, waveform_mode, encode_mode
    220         )
    221 

~\anaconda3\envs\Echopype\lib\site-packages\echopype\echodata\simrad.py in ?(echodata, waveform_mode, encode_mode)
    119     # initialize power and complex EchoData group values
    120     power_ed_group = None
    121     complex_ed_group = None
    122 
--> 123     transmit_type = echodata["Sonar/Beam_group1"]["transmit_type"]
    124     # assume transmit_type identical for all pings in a channel
    125     # TODO: change when allowing within-channel CW-BB switch
    126     first_ping_transmit_type = transmit_type.isel(ping_time=0)

~\anaconda3\envs\Echopype\lib\site-packages\xarray\core\dataset.py in ?(self, key)
   1540         """
   1541         if utils.is_dict_like(key):
   1542             return self.isel(**key)
   1543         if utils.hashable(key):
-> 1544             return self._construct_dataarray(key)
   1545         if utils.iterable_of_hashable(key):
   1546             return self._copy_listed(key)
   1547         raise ValueError(f"Unsupported key-type {type(key)}")

~\anaconda3\envs\Echopype\lib\site-packages\xarray\core\dataset.py in ?(self, name)
   1444 
   1445         try:
   1446             variable = self._variables[name]
   1447         except KeyError:
-> 1448             _, name, variable = _get_virtual_variable(self._variables, name, self.dims)
   1449 
   1450         needed_dims = set(variable.dims)
   1451 

~\anaconda3\envs\Echopype\lib\site-packages\xarray\core\dataset.py in ?(variables, key, dim_sizes)
    210         raise KeyError(key)
    211 
    212     split_key = key.split(".", 1)
    213     if len(split_key) != 2:
--> 214         raise KeyError(key)
    215 
    216     ref_name, var_name = split_key
    217     ref_var = variables[ref_name]

KeyError: 'transmit_type'

-----------------------------------------

Here the file I used which gets the error:
https://wetransfer.com/downloads/ee4a868b5db95171236132f6a6820b8b20231110080045/5d3a7e

Thanks a lot again for helping out!

praneethratna, 2023-11-11:
@icaro-00 Thanks for reporting this! could you share the .raw file instead so that we can debug this better?

icaro-00, 2023-11-12:
Hi @praneethratna! Here you can find the .raw file:

https://wetransfer.com/downloads/eb68d6d87000a25190be02c76d02863120231112054603/4a0e37

thanks!

icaro-00, 2023-11-12:
Hi @praneethratna,

herw you can find the .raw file:

https://wetransfer.com/downloads/eb68d6d87000a25190be02c76d02863120231112054603/4a0e37 

thanks for helping!

praneethratna, 2023-12-18:
Hey @icaro-00 Could you re-upload the file again if possible, so that we can look into the issue? Thanks

leewujung, 2024-04-10:
@icaro-00 : Also pinging here to see if you could provide an update of the link to download data files. Thanks!

leewujung, 2024-07-05:
Closing this as we do not have access to example files. @icaro-00 feel free to reopen.

closed 2024-07-05 by leewujung