echostack-org/echopype #1224: Allow multi-channel mask for `apply_mask` and simplify allowed `fill_value`

leewujung, 2023-11-13:
## Overview

Currently `mask.apply_mask` only accepts masks (a single mask or a list of masks) that do not have a `channel` dimension, and this mask will be applied to all channels. However, it would be more flexible for `mask.apply_mask` to be able to also accept masks that contain a channel dimension. For example for noise masks, it is likely that the mask for each channel would be different.

This does create the potentially problematic case where the `source_ds` and elements of the list of masks may have different channels. For example:
1. `mask=[mask1, mask2]`, where `mask1` has `channel=["chA"]` and `mask2` has `channel=["chB"]`
2. `source_ds` have `channel=["chA", "chB", "chC"]` (i.e., the combined masks would only have 2 of the 3 channels), so we need to decide what to do with `chC`

To handle the above:
1. Looks like doing [`xr.broadcast`](https://docs.xarray.dev/en/stable/generated/xarray.broadcast.html) before multiplying elements in the list of masks together would work (i.e., in the example above, the combined mask would have `channel=["chA", "chB"]`)
2. It seems more intuitive for channels that exist in `source_ds` but are missing in `mask` be left as is -- i.e. the mask would be all 1s for these channels. But perhaps it is the best to leave this as an option.

Also, looking at the code again, allowing `fill_value` to be of type `np.ndarray` would be really messy if we allow multi-channel masks, because there will be no way to match up which channel is which. I think we should just remove this and do `fill_value: Union[int, float, xr.DataArray] = np.nan`.

## Tasks
If we take the above approach, the tasks are:
- [ ] Remove `np.ndarray` from allowed type for `fill_value`
- [ ] If `fill_value` is of type `xr.DataArray`, it must have the same dimension as `source_ds`
- [ ] Check for the channel dimension for all elements in `mask` to ensure that they have the same shape in `ping_time` and `range_sample` (and `depth`, see #1204) -- this check should already be in place, but the code could use some checking and potentially simplifcation
- [ ] Make sure to broadcast all input masks together before combining them (element-wise multiplication)
- [ ] Add an input argument `keep_unmasked_channel: bool = True`
  - when `True`: channels that are not in `mask` will be left as is, i.e. no masking will be applied to the data in `chC`
  - when `False`: channels that are not in `mask` will be masked, i.e. all data in `chC` will be masked with values in `fill_value`.

@praneethratna @ctuguinay : Any thoughts?

ctuguinay, 2023-11-21:
This sounds good. I'll give this issue and #1204 a shot these next few days.

ctuguinay, 2024-04-09:
Closed with the merging of #1230

closed 2024-04-09 by ctuguinay