echostack-org/echopype #1239: [File conversion] No parser available for type MRU version 1

jmjech, 2023-12-04:
### General description of problem
_Required_
I have EK80 files collected within the past month using a Kongsberg WBT Tube that give me an error when I attempt to convert from .raw to .nc4. Data were collected on the uncrewed surface vehicle, Drix.

### Computing environment
_Required_
- Echopype version: 0.8.2 (installed on about 1 December 2023, but I did not uninstall a previous version)
- How echopype was installed (e.g., conda or pip): pip
- Operating system: Ubuntu 22.04.3 LTS

### Minimum example
_Required: A minimum code snippet that reproduces your problem._
The following code reproduces the errors I encountered:
```python
import echopype as ep
from echopype import open_raw
from pathlib import Path
filename = Path('/home/mjech/Desktop/WindDevelopment/EK80_Data/2023120_BIWF_WTG-3_Spiral-D20231021-T003735.raw')
filelist = [filename]
EKmodel = 'EK80'   #EK60
for f in filelist:
    ed = open_raw(str(f), sonar_model=EKmodel)

```

### Error message printouts
_Required: Copy-paste the entire error messages you encounter here._
Below is the error messages I received when running the above code:
```python
In [3]: %run Convert_EK80-to-netCDF.py
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
~/NOAA_Gdrive/sonarpros/Python_Programs/EK_ES/Convert_EK80-to-netCDF.py in <module>
     93 # convert each to netCDF4
     94 for f in filelist:
---> 95     ed = open_raw(str(f), sonar_model=EKmodel)
     96     # Henry B. Bigelow ICES code is 33HH
     97     ed['Platform']['platform_name'] = 'Drix18'

~/.local/lib/python3.10/site-packages/echopype/utils/prov.py in inner(*args, **kwargs)
    235             @functools.wraps(func)
    236             def inner(*args, **kwargs):
--> 237                 dataobj = func(*args, **kwargs)
    238                 if is_echodata:
    239                     ed = dataobj

~/.local/lib/python3.10/site-packages/echopype/convert/api.py in open_raw(raw_file, sonar_model, xml_path, convert_params, storage_options, use_swap, max_chunk_size)
    421     )
    422     # Actually parse the raw datagrams from source file
--> 423     parser.parse_raw()
    424 
    425     # Direct offload to zarr and rectangularization only available for some sonar models

~/.local/lib/python3.10/site-packages/echopype/convert/parse_base.py in parse_raw(self)
    367 
    368             # Read the rest of datagrams
--> 369             self._read_datagrams(fid)
    370 
    371         # Convert ping time to 1D numpy array, stored in dict indexed by channel,

~/.local/lib/python3.10/site-packages/echopype/convert/parse_base.py in _read_datagrams(self, fid)
    452                 #  when users only want CONFIG or ENV, but the way this is implemented
    453                 #  the raw0/3 datagrams are still parsed, you are just not saving them
--> 454                 new_datagram = fid.read(1)
    455 
    456             except SimradEOF:

~/.local/lib/python3.10/site-packages/echopype/convert/utils/ek_raw_io.py in read(self, k)
    434         if k == 1:
    435             try:
--> 436                 return self._read_next_dgram()
    437             except Exception:
    438                 if self.at_eof():

~/.local/lib/python3.10/site-packages/echopype/convert/utils/ek_raw_io.py in _read_next_dgram(self)
    344             return raw_dgram
    345         else:
--> 346             nice_dgram = self._convert_raw_datagram(raw_dgram, bytes_read)
    347             self._current_dgram_offset += 1
    348             return nice_dgram

~/.local/lib/python3.10/site-packages/echopype/convert/utils/ek_raw_io.py in _convert_raw_datagram(self, raw_datagram_string, bytes_read)
    373             return raw_datagram_string
    374 
--> 375         nice_dgram = parser.from_string(raw_datagram_string, bytes_read)
    376         return nice_dgram
    377 

~/.local/lib/python3.10/site-packages/echopype/convert/utils/ek_raw_parsers.py in from_string(self, raw_string, bytes_read)
     78         if sys.version_info.major > 2:
     79             header = header.decode()
---> 80         id_, version = self.validate_data_header(header)
     81         return self._unpack_contents(raw_string, bytes_read, version=version)
     82 

~/.local/lib/python3.10/site-packages/echopype/convert/utils/ek_raw_parsers.py in validate_data_header(self, data)
     70 
     71         if version not in self._versions:
---> 72             raise ValueError("No parser available for type %s version %d" % (self._id, version))
     73 
     74         return type_, version

ValueError: No parser available for type MRU version 1

```

### Provide an example file
_Required: An example file that allow reproducing the problem._
- Link to an example raw file: SOME_ONLINE_LOCATION. Smaller files are strongly preferred (e.g., <25MB for AZFP and EK60, <100MB for EK80).
- Raw file sensor model: (FOR EXAMPLE: Simrad EK60)

Please e-mail for a link to the data file, ~200 MB. michael.jech@noaa.gov

### Troubleshooting
_Optional but preferred: Explain any steps that you have taken to identify the cause of conversion error._
I attempted to follow the different programs/scripts, but it seems to be some missing capability.

leewujung, 2023-12-05:
Thanks @jmjech ! We'll add this to the parser. 

Ref: https://www.simrad.online/ek80/interface/ek80_interface_en_us.pdf#page=162

jmjech, 2023-12-14:
I e-mailed a 2MB file to Wu-Jung. Let me know if it works for you. I have a few other "smaller" files.

closed 2023-12-18 by praneethratna