echostack-org/echopype #1256: [File conversion] SUSPECT_CAUSE

ZHANGGU, 2024-01-28:
### General description of problem
_Required_
I cannot convert AZFP data
### Computing environment
_Required_
- Echopype version:echopype-0.8.3
- How echopype was installed (e.g., conda or pip): WinPython
- Operating system: windows

### Minimum example
_Required: A minimum code snippet that reproduces your problem._
The following code reproduces the errors I encountered:
```python
CODE SNIPPET
```
import echopype as ep
import os
from datetime import datetime

src_dir = r'G:\2023200002-ANTARCTIC-PROVIDER\DATA-INNKOMMENDE\HYDROAKUSTIKK-ASL-AZFP\DATA'
dst_dir = r'E:\MyWork\Krill\2023200002-ANTARCTIC-PROVIDER\ASL_AZFP'

xml_fp = r'G:\2023200002-ANTARCTIC-PROVIDER\DATA-INNKOMMENDE\HYDROAKUSTIKK-ASL-AZFP\DATA\202002\20021418.XML'
post_fix = '01A'
for c in os.listdir(src_dir):
    sub_c = os.path.join(src_dir, c)
    for f in os.listdir(sub_c):
        if f.find(post_fix)>0:
            fp = os.path.join(sub_c, f)
            fmtime = os.path.getmtime(fp)
            mo_datetime = datetime.fromtimestamp(fmtime)
            ftimestamp = datetime.strftime(mo_datetime, '%Y%m%%dT%H%M%S')
            ed = ep.open_raw(fp, sonar_model='AZFP', xml_path=xml_fp)
            nc_fp = os.path.join(dst_dir, f.replace(post_fix, '{:}.nc'.format(ftimestamp)))
            ed.to_netcdf(save_path=nc_fp)
            print('{:} converted'.format(f))
### Error message printouts
_Required: Copy-paste the entire error messages you encounter here._
Below is the error messages I received when running the above code:
```python
ERROR MESSAGES
```

TypeError: 'float' object is not subscriptable

### Provide an example file
_Required: An example file that allow reproducing the problem._
- Link to an example raw file: SOME_ONLINE_LOCATION. Smaller files are strongly preferred (e.g., <25MB for AZFP and EK60, <100MB for EK80).
- Raw file sensor model: (FOR EXAMPLE: Simrad EK60)

### Troubleshooting
_Optional but preferred: Explain any steps that you have taken to identify the cause of conversion error._

This is a float that has no reference [freq_ch]
    v = (np.log10(v) - 2.5) * (8 * 65535) * self.parameters["DS"][freq_ch]
[image]

praneethratna, 2024-02-03:
@ZHANGGU Thanks for reporting this! could you share the data files so that we can debug this better?

ZHANGGU, 2024-02-03:
I tried to correct the bug but some other bugs came out as well. It cannot read my AZFP file of the  version 01A
[Uploading 20021600.zip…]()
. I uploaded a sample file. would you plz share the data specification?

praneethratna, 2024-02-03:
@ZHANGGU The link points to github issue, could you upload the data files and share a link to same here?

ZHANGGU, 2024-02-03:
[Uploading 20021600.zip…]()

leewujung, 2024-02-25:
@ZHANGGU : Could you provide a new link to an example file? The previous link does not point to a file.

ctuguinay, 2024-03-30:
Hi @ZHANGGU, in your zip above, you only included the `.01A` files, but the parsing of AZFP needs `.XML` files too. Could you resend a zip with the `.01A` and `.XML` files?

leewujung, 2024-04-18:
Hey @ZHANGGU : Could you provide an XML file for this so that we can look into the issue?

ZHANGGU, 2024-04-18:
[14021116.zip](https://github.com/OSOceanAcoustics/echopype/files/15029224/14021116.zip)
@leewujung uploaded

ctuguinay, 2024-04-18:
@ZHANGGU

So @leewujung  and I looked at your file and we saw that there are two sections here that imply different things about the frequency: The first section implies that there is 1 frequency in the DS and it's stated to be 125kHz, and the second section implies that there are 4 frequencies. Does your current setup have 4 125 kHz channels?

ZHANGGU, 2024-04-19:
only 125 kHz was used. Azfp link can export without problems

leewujung, 2024-04-19:
@ZHANGGU : We are not part of ASL and don't have access to AZFP link source code.

@ctuguinay : The only place I can see the association between the active channels and the parameters is using the entries under AZFP_Parameters/Header/Boards:
This is from the provided file:
[image]
This is from one of our test files:
[image]
But it's not clear to me how this is mapped to the info under Phases that tells the instrument when to transmit.
Let's connect with ASL to find out.

ZHANGGU, 2024-04-19:
Azfp supports 4 frequencies but we did configured only 1 frequency of 125 kHz!  It does not need to use 4 frequencies in practice. The code should reflect practices. thx