echostack-org/echopype #1263: Upgrade handling of test data files to use GitHub Release Assets

leewujung, 2024-02-25:
Right now test data files are stored in a private google drive folder and copied to the virtual machine when running tests. This is somewhat brittle and makes this project less open to contributions. 

@ocefpaf suggested the option to use GitHub Release Assets to store test data files, which keeps the data in the same repo but does not contaminate the code, with the additional advantage that the test data can then be versioned. Since none of our test data files >2G so will not cause problem with the assets, this should work pretty well.

@lsetiawan has given this a try here [here](https://github.com/lsetiawan/echopype-test-data/releases), so seems indeed we can go this route to handle our test data files.

dextrot, 2024-02-26:
Hi @leewujung 
I want to work on this issue, can you please assign it to me?

HQ786, 2024-03-15:
Hi @leewujung
I am interested to work on this issue, can you please assign this to me?

oftfrfbf, 2024-11-25:
@leewujung  just pinging this issue to say that I am currently working on addressing the testing resource limitations. I have my development environment set up and am currently researching other projects use of the _pooch_ utility mentioned to see if it will work here.

> @ocefpaf suggested the option to use GitHub Release Assets to store test data files

With respect to the "github release assets." Is the expected workflow to create a consolidated release of test_data files embedded within the echopype codebase as git LFS objects? I am not entirely familiar with the process yet, the best I can interpret is that releases can contain LFS objects as mentioned here:

https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/managing-repository-settings/managing-git-lfs-objects-in-archives-of-your-repository

So when tagging a release, such as "v0.9.1," this could conceivably include all the 2.95 GB of needed test_data assets as LFS objects.

Anyway, I will be working on a prototype to see if I can consolidate files so we don't need google-drive and docker integration to serve test files.

ocefpaf, 2024-11-26:
@oftfrfbf here is an example of what I do: https://github.com/pyoceans/pocean-core/releases

You can see two releases, [one for data](https://github.com/pyoceans/pocean-core/releases/tag/2024.04) and one for the code. The data release uses CalVer to avoid clashing with the library that uses SemVer. All the data files are added as assets to the release and fetched/cached at test time [using pooch](https://github.com/pyoceans/pocean-core/blob/main/pocean/tests/download_test_data.py). Let me know if you have any questions.

Note that, while I prefer to keep them both in the same repository, one can have a repo for data and another one for the code.

oftfrfbf, 2024-12-03:
Thanks @ocefpaf, I had been interpreting the goals wrong previously and this helps point me in the right direction! I will be working on a prototype...

oftfrfbf, 2024-12-23:
Just wanted to share an update. I have been working on this feature here:

https://github.com/oftfrfbf/echopype/tree/EP-1263

and am pretty close to having all the functionality implemented. I have really been enjoying getting to learn pooch!

To start off with, I have created a second repo here, https://github.com/oftfrfbf/echopype-test-data, with test assets:

https://github.com/oftfrfbf/echopype-test-data/releases/tag/2024.12.23.10.10

Having a secondary github repo isn't required, so I would suggest just creating tagged releases within the echopype project for the sake of simplicity.

One thing to note is that release assets require a tag to be assoicated with the files. Also of note, those assets do not add to the repos overall git size (which is great!), the only requirement is that any individual asset is less than 2 GB when uploading. I opted to zip each respective folder and add all 18 as assets. I also calculated a sha256 sum for each respective zip and added that to the description when uploading.

Back in the conftest.py for echopype I can create a "registry" that associates each zip file with a hash and the tagged release URL, (see here for more <https://github.com/oftfrfbf/echopype/blob/main/echopype/tests/conftest.py>) e.g.:
```
RELEASE_ASSET_VERSION = '2024.12.23.10.10'

ECHOPYPE_RESOURCES = pooch.create(
    path=pooch.os_cache("echopype"),
    base_url="https://github.com/oftfrfbf/echopype-test-data/releases/download/{version}/",
    version=RELEASE_ASSET_VERSION,
    retry_if_failed=1,
    registry={
        "ad2cp.zip": "sha256:8c0e45451eca31b478e7ba9d265fc1bb5257045d30dc50fc5299d2df2abe8430",
        "azfp.zip": "sha256:6b2067e18e71e7752b768cb84284fef3e0d82b5b775ea2499d52df8202936415",
        "azfp6.zip": "sha256:b13ad0cb026d42bd0112d2999e5f63ba28226e4c79ffe335d650fe3f28db760d",
        "ea640.zip": "sha256:10bcd57c9d382e680e26a0f78b3e1c8bda8c68799d69334ab63031c10650d114",
        "ecs.zip": "sha256:e873754f40ec7142c5ece9706a9c63e6f49666b534c79aeac54952ece8267439",
        "ek60.zip": "sha256:872e8c021dc8b2fe22b8ff62659c1737e80bddee5835429d20e51a7a71dd9ef8",
        "ek60_missing_channel_power.zip": "sha256:82a9f89c848f6925f779ef6b8d47e6fb1c59e720a303830010f73e45e82c6609",
        "ek80.zip": "sha256:e46dea3ba4531d2576bbae2cd17a33ead48cb594244dd611fddd98e53901aa39",
        "ek80_bb_complex_multiplex.zip": "sha256:4eac7b0f40bdd8405aac38e114b47cc1901f7614002e7a5ea9a642bbcb884f93",
        "ek80_bb_with_calibration.zip": "sha256:362613ecce383136be16e0a7d7c74a33ef11866d5a0b6bf1581172b27186790a",
        "ek80_duplicate_ping_times.zip": "sha256:f2affc25c0972c23ef2f9e1e78a7627bffe6105d39e5b4e9c68b3e8bb81e08b5",
        "ek80_ext.zip": "sha256:dbaf35f2af1526e4d5a9e6b65b22e090d4db0e0355d8be8e5c2255fe03b65475",
        "ek80_invalid_env_datagrams.zip": "sha256:12f8bc32aecaf3c1c8a506491be2346382aed76f6c6dd8d60847f6915b597c50",
        "ek80_missing_sound_velocity_profile.zip": "sha256:56fd98974cf1dd1cd400358045cfbcf82c52e431e4acf822146458dbdf11a59e",
        "ek80_new.zip": "sha256:644ef2c2d89e52b63635eaf523eb0b0385580b1bacd189aa39c62db69943abb6",
        "es60.zip": "sha256:7abf0635a7d1365d075d9b4ce32abef86efe2ea1fda84be4f07187f384709ff5",
        "es70.zip": "sha256:e7d58335b0049b3709e08663913925d75a76250f787f84019dc5f3fadfc0983a",
        "es80.zip": "sha256:9547583dbd6ac77375384e614cd559ff61639d36f082f1ce80855f9b57a52213",
    },
)
```

The "pooch.os_cache("echopype")" creates a local cache directory where downloaded files get stored. In a windows system this is at a location such as 

>C:\Users\<user>\AppData\Local\echopype\Cache\2024.12.23.10.10

Or

>/Users/<user>/Library/Caches/echopype/2024.12.23.10.10

for macOS.

From there I have a couple helper functions for downloading the files to the cache directory, verifying checksums, unzipping, and wiring the data back into each of the respective test_path pytest session fixtures so that test resources can be referenced seamlessly in tests.

I have spent a lot of time getting rid of references to 'echopype/test_data' throughout the code and instead use fixtures such as:
```python
@pytest.fixture
def ek60_path(test_path):
    return test_path['EK60']
```
to establish paths to test files. This is working really well so far!

@leewujung, one thing that I am hoping is okay is if the "echopype/test_data" folder can disappear completely? We can now establish these cache directories instead where the data will automatically download/cache/increment as successive versions of the data are published. Also worth noting is that I have rewritten the 'dump' directory to write to that same cache location. 

One thing I still need to work on is ensuring that the github build/test actions will handle this new feature properly so I will see what the best way to approach that is. The github actions do not work for forked repos as far as I have seen and so I've turned them off for my copy.

Overall I think this is a great library and will really help consolidate the test files to live alongside the code while also allowing for opensource contributions.

Moving forward I will comment out the "httpserver" from the docker-compose.yaml and test that everything works properly for ci/cd builds. The next step would be for @leewujung to create the release assets in echopype so I can update the base_url & version in the registry. I'll try to create a pull request in the next couple days but also realize it's holiday time for many. Hope everyone is having a nice break!

leewujung, 2025-07-28:
Hey @oftfrfbf: @dbashford-NOAA is getting up to speed with echopype dev now, and I think we could probably get to this in a few weeks (need to upgrade to zarr>3 first...). Do you think you could help him pick up from where you have left off for the release assets then? 

I also finally had a chance to read through what you wrote above - I can create the release assets once we get to this.

Thanks!

LOCEANlloydizard, 2025-08-08:
Hi all! Apologies, I hadn’t seen this issue earlier (should’ve searched first 😅). I’ve recently been exploring GitHub Release Assets as a way to work around the Docker limitations on Windows runners for testing, and just now came across this thread.

In case it’s helpful: I managed to get a Windows runner working in my fork using GitHub releases to load the test data, and shared the approach in [#963](https://github.com/OSOceanAcoustics/echopype/pull/963). The implementation doesn’t use pooch, so it’s really great to see this cleaner and more sustainable solution taking shape here! I wasn’t familiar with the package, it looks super useful.

Also, regarding GitHub Actions not working on forks: I tested the workflow on my fork with two jobs (one per OS), and it seems to behave as expected 🙂 Just thought I’d mention it in case it helps move this forward!

cheers!

LOCEANlloydizard, 2025-10-20:
Hi all,
This is now proposed in PR #1562, based on the previous comments and particularly the work from @ocefpaf and @oftfrfbf!
Cheers!