echostack-org/echopype #1264: I can't plot my EK80 files using epviz.create_echogram

bluesteven, 2024-02-26:
source codes is as following:

import echopype as ep

raw_url = "D:/DwnlData/echotype/allek80_200k/D20240129-T031539.raw"
echodata = ep.open_raw(raw_file=raw_url, sonar_model='EK80')

# epviz.create_echogram(echodata, range_kwargs={'waveform_mode':'BB', 'encode_mode':"complex"})
echodata.to_netcdf(save_path='./unpacked_files')
# Compute volume backscattering strength (Sv) from raw data
# ds_Sv = ep.calibrate.compute_Sv(echodata)
ds_Sv = ep.calibrate.compute_Sv(echodata,waveform_mode='BB', encode_mode='complex')

import echopype.visualize as epviz
# Quickly look at all of the frequency, calculate range on the fly, and color the data based on the actual data range available.
epviz.create_echogram(ds_Sv, get_range=True, robust=True)

I get these information and error

D:\DwnlData\echotype\.venv\lib\site-packages\echopype\utils\coding.py:87: UserWarning: Times can't be serialized faithfully to int64 with requested units 'seconds since 1900-01-01T00:00:00+00:00'. Resolution of 'microseconds' needed. Serializing times to floating point instead. Set encoding['dtype'] to integer dtype to serialize to int64. Set encoding['dtype'] to floating point dtype to silence this warning.
  encoded_data, _, _ = coding.times.encode_cf_datetime(

UFuncTypeError                            Traceback (most recent call last)
Cell In[3], line 14
     12 import echopype.visualize as epviz
     13 # Quickly look at all of the frequency, calculate range on the fly, and color the data based on the actual data range available.
---> 14 epviz.create_echogram(ds_Sv, get_range=True, robust=True)

File D:\DwnlData\echotype\.venv\lib\site-packages\echopype\visualize\api.py:206, in create_echogram(data, channel, frequency, get_range, range_kwargs, vertical_offset, **kwargs)
    203 else:
    204     raise ValueError(f"Unsupported data type: {type(data)}")
--> 206 return _plot_echogram(
    207     ds,
    208     xaxis='ping_time',
    209     yaxis=yaxis,
    210     variable=variable,
    211     channel=channel,
    212     frequency=frequency,
    213     **kwargs,
    214 )

File D:\DwnlData\echotype\.venv\lib\site-packages\echopype\visualize\plot.py:258, in _plot_echogram(ds, channel, frequency, variable, xaxis, yaxis, **kwargs)
    251     d = d.sel(
    252         ping_time=d.echo_range.dropna(dim='ping_time', how='all').ping_time
    253     )
    254     d = d.sel(
    255         range_sample=d.echo_range.dropna(dim='range_sample').range_sample
    256     )
--> 258 plot = d.plot.pcolormesh(
    259     x=xaxis,
    260     y=yaxis,
    261     col=col,
    262     row=row,
    263     **kwargs,
    264 )
    265 _set_label(plot, filter_var=filter_var, filter_val=filter_val, col=col)
    266 plots.append(plot)

File D:\DwnlData\echotype\.venv\lib\site-packages\xarray\plot\accessor.py:784, in DataArrayPlotAccessor.pcolormesh(self, *args, **kwargs)
    782 @functools.wraps(dataarray_plot.pcolormesh, assigned=("__doc__",))
    783 def pcolormesh(self, *args, **kwargs) -> QuadMesh | FacetGrid[DataArray]:
--> 784     return dataarray_plot.pcolormesh(self._da, *args, **kwargs)

File D:\DwnlData\echotype\.venv\lib\site-packages\xarray\plot\dataarray_plot.py:1601, in _plot2d.<locals>.newplotfunc(***failed resolving arguments***)
   1597     raise ValueError("plt.imshow's `aspect` kwarg is not available in xarray")
   1599 ax = get_axis(figsize, size, aspect, ax, **subplot_kws)
-> 1601 primitive = plotfunc(
   1602     xplt,
   1603     yplt,
   1604     zval,
   1605     ax=ax,
   1606     cmap=cmap_params["cmap"],
   1607     vmin=cmap_params["vmin"],
   1608     vmax=cmap_params["vmax"],
   1609     norm=cmap_params["norm"],
   1610     **kwargs,
   1611 )
   1613 # Label the plot with metadata
   1614 if add_labels:

File D:\DwnlData\echotype\.venv\lib\site-packages\xarray\plot\dataarray_plot.py:2298, in pcolormesh(x, y, z, ax, xscale, yscale, infer_intervals, **kwargs)
   2295     else:
   2296         # we have to infer the intervals on both axes
   2297         x = _infer_interval_breaks(x, axis=1, scale=xscale)
-> 2298         x = _infer_interval_breaks(x, axis=0, scale=xscale)
   2300 if infer_intervals and (np.shape(y)[0] == np.shape(z)[0]):
   2301     if y.ndim == 1:

File D:\DwnlData\echotype\.venv\lib\site-packages\xarray\plot\utils.py:868, in _infer_interval_breaks(coord, axis, scale, check_monotonic)
    866 if deltas.size == 0:
    867     deltas = np.array(0.0)
--> 868 first = np.take(coord, [0], axis=axis) - np.take(deltas, [0], axis=axis)
    869 last = np.take(coord, [-1], axis=axis) + np.take(deltas, [-1], axis=axis)
    870 trim_last = tuple(
    871     slice(None, -1) if n == axis else slice(None) for n in range(coord.ndim)
    872 )

UFuncTypeError: ufunc 'subtract' cannot use operands with types dtype('<M8[ns]') and dtype('float64')

leewujung, 2024-03-10:
@bluesteven : I can't tell from your error message what is going on. The step that errored out is within xarray, so it is possibly some problem with the Sv data.

What do you get by using `ds_Sv["Sv"].isel(channel=0).plot(x="ping_time")`?
This command should plot the Sv data of the first channel.

You can also try using `matplotlib` just to see if the Sv data would plot without the whistles and bells added by xarray that uses the dimensions.
```python
import matplotlib.pyplot as plt
plt.imshow(ds_Sv["Sv"].isel(channel=0).values)   # .values returns a numpy array
```

Also FYI - we are considering retiring the `epviz` subpackage (see #1276).

leewujung, 2024-06-15:
The `visualize` package will be removed as of the upcoming v0.9.0 release. I'll close this after that.

leewujung, 2024-07-02:
Closed via #1336.

closed 2024-07-02 by leewujung