echostack-org/echopype #1276: Retire the `visualize` subpackage?

leewujung, 2024-03-10:
I am wondering if we should retire the subpackage `echopype.visualize` (`epviz`), and keep the EK500 colormap import. [Current docs here](https://echopype.readthedocs.io/en/stable/viz.html).

There are a few main reasons:
- The current epviz functionality tries to guess automatically what users want to do if they miss some input arguments, but the handling is brittle if the data are not in one of the predicted form. Given the flexibility of EK80 configuration, the "guess" often breaks.
- It seems that instead of having a specialized subpackage, we can just do a better job in the docs to show users how to plot data using native xarray plotting functions. This way users will likely be able to explore it in a more flexible way.
- Now we have the [`Echoshader`](https://github.com/OSOceanAcoustics/echoshader) package that provides interactive visualization functionalities via the accessor mechanism that is more powerful for the users, so we can direct users there.

@lsetiawan: Any thoughts? I know you spent quite a bit of time developing epviz and writing the docs too.

lsetiawan, 2024-03-15:
Now that all the plotting functionalities are mostly within Echoshader, I agree that this `visualize` subpackage can be retired. If there's a way to port the EK500 colormap that would be really great!

> It seems that instead of having a specialized subpackage, we can just do a better job in the docs to show users how to plot data using native xarray plotting functions.

I really like this idea of showing users how to use xarray plotting functions to build their own static plots!

leewujung, 2024-03-15:
Ok, I'll put in a deprecating warning for v0.8.4 and redirect people to Echoshader, and remove it at v0.9.0 (since its current state is buggy and will be very involved to fix).

leewujung, 2024-07-05:
This can be closed now via #1347.

closed 2024-07-05 by leewujung