echostack-org/echopype #1288: [GSoC24] ...

huan24080, 2024-03-23:
### Discussion topic

Introducing the "security-and-quality" Query Suite 
Abstract:
The "security-and-quality" Query Suite is an essential addition to GitHub's CodeQL query suites, providing a comprehensive approach to identifying both security vulnerabilities and code quality issues within codebases. This project aims to introduce and integrate the "security-and-quality" Query Suite into GitHub's code scanning configuration, enhancing developers' ability to proactively detect and address potential risks and inefficiencies in their software projects.

Why this project?
The integration of the "security-and-quality" Query Suite addresses a critical need in modern software development practices. By combining security-focused analysis with code quality assessments, developers can ensure that their code not only meets security standards but also adheres to best practices for maintainability and scalability. I am motivated to undertake this project as it aligns with my passion for improving software quality and security, and I believe it will have a significant impact on the developer community.

Technical Details:
This project will involve:

Utilizing GitHub's CodeQL infrastructure and APIs for integrating the "security-and-quality" Query Suite into the code scanning configuration.
Implementing query rules and checks to cover a wide range of security vulnerabilities (e.g., SQL injection, cross-site scripting) and code quality issues (e.g., code duplication, complexity).
Collaborating with mentors to refine and optimize query performance and accuracy.
Engaging with the GitHub developer community for feedback and contributions to enhance the effectiveness of the Query Suite.
Development Experience:
I have prior experience contributing to open-source projects on GitHub and have successfully completed university courses in software engineering and cybersecurity. My GitHub profile showcases my contributions, and I have received positive feedback on my code quality and collaboration skills. I am confident in my ability to navigate complex codebases and implement effective solutions.

### Related existing issues or PRs

https://github.com/github/docs/issues/32160

huan24080, 2024-03-23:
I want to make a proposal based on this is it possible ?

leewujung, 2024-04-01:
@huan24080 : Thanks for raising the discussion topic! Could you provide more details on which part of the package you think this enhancement will be useful, and how that addresses our GSoC24 goals on upgrading testing suite and scalability when handling large dataset?

closed 2024-05-01 by leewujung