echostack-org/echopype #1290: Serialization warning on timestamp resolution

leewujung, 2024-03-28:
This came from #1232:

There is a serialization warning with timestamp resolution. This seems something that came up after pandas 2.0 update in xarray. It would be good to resolve this, since if not suppressed, a huge bulk of warnings would show up when mass-converting files. It is also good to make sure we know if there is any resolution issue.

Sample file:
[SCAN-D20231122-T172241.zip](https://github.com/OSOceanAcoustics/echopype/files/13457137/SCAN-D20231122-T172241.zip)

Warning message:
```
/lib/python3.9/site-packages/echopype/utils/coding.py:87: 
UserWarning: Times can't be serialized faithfully to int64 with requested units 'seconds since 1900-01-01T00:00:00+00:00'. 
Resolution of 'microseconds' needed. 
Serializing times to floating point instead. 
Set encoding['dtype'] to integer dtype to serialize to int64. 
Set encoding['dtype'] to floating point dtype to silence this warning.
  encoded_data, _, _ = coding.times.encode_cf_datetime(
```

ctuguinay, 2024-04-01:
* Look through issues in Xarray to see if this is addressed and if not raise an issue
* Just filter it out for now

ctuguinay, 2024-04-03:
@leewujung 
So this warning pops up here where xarray enforces that if the `dtype` passed in is `None` then a 'not faithful' serialization warning is passed in: https://github.com/pydata/xarray/blob/97d3a3aaa071fa5341132331abe90ec39f914b52/xarray/coding/times.py#L772C1-L778C18. Echopype calls the function that this code is in in `_encode_dataarray`: https://github.com/OSOceanAcoustics/echopype/blob/4278dd6d67389d04aa0a4f28f282d78c5f823a4e/echopype/utils/coding.py#L74C1-L91C74

This change stems from this PR that was created to fix a series of cf datetime time issues that arose with Pandas 2.0: https://github.com/pydata/xarray/commit/b6d4bf72fd56014c0bd142bc438d6204d8dfd1ef

In this gist, I try out a few dtype and unit permutations with nanosecond resolution data, just to see what issues and errors they bring up: https://gist.github.com/ctuguinay/c77461f81ee8229ea43ba2da33b7f83c. Even in the case where no warning or error is raised (when encode kwaargs are "units": "seconds since 1900-01-01T00:00:00+00:00", "calendar": "gregorian", "dtype": np.dtype("float64")), we are still met with nanosecond resolution difference between encoded input and decoded output. I'm not sure if this is what is intended for `_encode_dataarray`.

ctuguinay, 2024-04-17:
Closed by #1299

closed 2024-04-17 by ctuguinay